chipsalliance/chisel #262: black box instance names aren't taken from Chisel code

mwachs5, 2016-08-19:
For most modules, the emitted verilog is now very meaningful (thanks!). However, this is not true for modules which are black boxes. An example is SimDTM in TestHarness.scala. It has a name (dtm)in the Chisel but in the final Verilog it is SimDTM. 

class BlackBoxModule extends BlackBox {
...
}

class MyModule extends Module {

...

val my_black_box = Module ( new BlackBoxModule)

...

}

This elaborates with an instance name of BlackBoxModule_1 instead my_black_box.

aswaterman, 2016-08-19:
I took a quick look and it's not immediately obvious why this is going wrong.  I'd like a fix, too!

kammoh, 2016-08-19:
strange... I can't seam to reproduce this
```
  module MyModule : 
    input clk : Clock
    input reset : UInt<1>
    output io : {}
    
    io is invalid
    inst my_black_box of BlackBoxModule @[Test.scala 13:29] 
````

mwachs5, 2016-08-19:
I think that actually it has nothing to do with black box. Is this expected
behavior if the module or black box is instantiated conditionally (inside
an if())?

kammoh, 2016-08-19:
This also produces the correct instance name (my_gray_module) for me:
```
class BlackBoxModule extends BlackBox {
  val io = new Bundle {
    val in = UInt(INPUT, width = 8)
  }

}

class SubModule extends BlackBox {
  val io = new Bundle {
    val in = UInt(INPUT, width = 8)
  }
}

class MyModule(val black: Boolean) extends Module {
  val io = new Bundle {
    val in = UInt(INPUT, width = 8)
  }

  val my_gray_module = Module(
    if (black)
      new BlackBoxModule
    else
      new SubModule
  )
}
```
Am I missing something here?

mwachs5, 2016-08-19:
That's not exactly what I meant.  I'll create a small test case that actually shows my question.

mwachs5, 2016-08-19:
Here's what I'm actually seeing, and understand if it's a less simple issue to resolve. I'll update the title of this issue. It is if you conditionally declare the val at all, not just decide what to assign to it conditionally. 

```
class Simple extends Module {
  val io = new Bundle {
    val a = Bool(INPUT)
    val b = Bool(OUTPUT)
  }
  b := a
}

class Top extends Module {

  if (true) {
    val my_simple = Module (new Simple())
  }
}

```

For me, this produces:
```

  Simple Simple_1 (
    .clk(Simple_1_clk),
    .reset(Simple_1_reset),
    .io_a(Simple_1_io_a),
    .io_b(Simple_1_io_b)
  );

```

kammoh, 2016-08-19:
That's expected behavior as the current naming mechanism uses scala(java) reflection to name the members. It only looks for the "val"s inside the body of your Module class (or the classes you may have extended from). 
I believe we can enhance this naming mechanism inside the Module() macro by checking the parent context tree. AFAIR sbt uses the same mechanism for its "setting" key namings.

mwachs5, 2016-08-19:
Cool, that all makes sense. I was trying to figure out how to flag this as "feature request" vs actual bug/issue, but not sure how.

ducky64, 2016-11-21:
Superseded by #303 and implementation pending approval in #327

closed 2016-11-21 by ducky64