chipsalliance/chisel #263: A way of tagging Chisel outputs with additional info (?) -- Re: memories

shunshou, 2016-08-21:
So, Firrtl seems to have "Info" fields. Is there some existing way to tag additional info at the Chisel level to have it accessible in Firrtl?

My particular scenario (pertaining to memories):

If required memory cannot be supported by a memory compiler, it's obvious that it needs to be split into smaller chunks.

However, if memory can be supported by a memory compiler, it's not immediately obvious that it should not be split. Specifically, if you look at area/power/timing, you'll see that, generally, the overhead of peripheral circuits (bit line sense amps, etc.) makes it a bad idea to split memory. Theoretically, the memory compilers have been optimized to generated the ideal solution. *However* one major thing that this doesn't account for is external routing. The SRAM macros placed in your design are essentially huge chunks of routing blockages for the rest of your design, which can be pretty bad depending on where the SRAMs are located. As I've learned recently from people with more experience, typically, if you expect the SRAMs to be placed somewhere in the center of your chip (L1?), you want to split large SRAMs into smaller chunks, whereas SRAMs placed on chip edges (L2?) are better off not being split. This isn't really justified (can't look only at generated SRAM #'s for optimization) until you get to the floorplanning stages of chip design.

When you're designing something like a processor, you know ahead of time what Chisel memories are L1, L2, etc. and *might* already have some idea of where they're typically placed on a chip. It'd be great if you could (not a requirement) choose to annotate this in some way and have it propagated into Firrtl so certain passes may or may not use them to perform optimizations.

Current setup: Chisel/Firrtl emits a generic black box wrapper + conf file that needs to go through vlsi_mem_gen. vlsi_mem_gen isn't particularly smart, so it only picks out valid memory sizes from memory compilers, and if memories need to be split b/c there are no valid sizes, vlsi_mem_gen (or similar) needs to split memories and re-connect them up appropriately. 

Up until now, I don't think anyone's really tried to push memory/chip performance, and that's why this has been a "good enough" solution. 

Ideal setup: Absorb vlsi_mem_gen functionality into a set of custom Firrtl transforms with input configuration information.

Input conf information could be:
- Valid macro sizes, their area/performance to solve for some optimal configuration (ideal for "edge" SRAMs)
- The above + loose floorplanning info associated with each generated SRAM (edge, center, etc.)

Then, for me, the question becomes -- should the floorplanning info be part of a separate conf file or should it just be optionally annotated in Chisel? If it's a separate Firrtl pass conf file, you'd have to match a generated DefMemory name to this info, and it's a pretty manual process (esp. if the memory names are currently something like T_406 for *cache).

Anyways, if the Firrtl pass already had some idea about floorplanning, it could be smarter in determining whether memories should be split or not, and you wouldn't need to separately write something new that would essentially break established connections to do more optimization after Firrtl has emitted Verilog. 

I know people want default Rocket to be "pure", and in that sense, maybe no annotation should be given "i.e. tagged with NoInfo", and then the pass could just not perform this optimization. I assume people building on top of Rocket will have their own parameter customizations (via CDE?) that could then override these defaults. 

And ideally, the pass would be able to generate a conf file that would aid in floorplan tcl-scripting, because right now, there's not a really good bridge between RTL design and the engineering required to make the RTL live up to its full potential (i.e. there's so much to be done at the layout level, and smarter decisions can only be made with more info about RTL. Right now, this info is pretty much all just communicated verbally, but it doesn't need to be that way... ==> my interpretation of "agile (but still great QoR) hardware"). 

@azidar @yunsup @aswaterman @jackkoenig -- thoughts? lol

azidar, 2016-08-22:
So you are suggesting using the "info" field to pass annotations from Chisel3 to a custom Firrtl pass. Instead, you should be using the annotation system which is designed specifically for this use case. It's in its early stages so getting more use cases would be very helpful. All of what you are saying to generate a conf file, have that consumed by downstream passes etc. makes total sense and is a great idea.

> Edit: Also, since this is an extremely specific pass, I don't think it should be part of default Firrtl. It should be some utility ASICFirrtl pass that can be run on if you so choose. That way, base Firrtl can be "pure", but you can leverage pass-writing to enhance QoR.

Right. We should think about writing libraries, not tools. I'd be happy to add some of these passes as a library that works with core Firrtl. Similarly, the annotation API is in core Firrtl, but specific instances like this system shouldn't be.

ducky64, 2017-02-06:
Solved with annotations.

closed 2017-02-06 by ducky64

colinschmidt, 2017-02-06:
For example you can add additional pins with some annotations to firrtl.

shunshou, 2017-02-06:
Is there an example? Say, for AnalogTypes, I want to attach an Analog inout to a pad, but I want to annotate (in Chisel or at some higher level) if the pad required is "slow" or "fast"... and use that as part of a FIRRTL transform.