chipsalliance/chisel #264: Chisel3 needs to provide an Annotation API

chick, 2016-08-22:
Chisel3 should provide API for generating and passing annotations to Firrtl passes.  With such an API, development of passes will be difficult and will result in multiple, likely conflicting, implementations.

azidar, 2016-08-22:
see #263 for use case

shunshou, 2016-08-25:
Another use case for annotating black boxes: Based off of conversations with other people, having black boxes annotated with things like "isPLL" etc. would be useful for separate backends to match corresponding Verilog (FPGA/ASIC specific). Or maybe can you create a library of parts that extend BlackBoxes like PLL extends BlackBox, etc. with appropriate annotation info that can then be used for configuration at the backend?

chick, 2016-12-15:
The original intent of this issue has been completed with PR #405 
@shunshou I don't think this addresses your question.  I believe that the chisel annotation system will allow you to annotate blackboxes at least to the point of getting those annotations to a custom firrtl pass, but what happens after that is not clear.  Perhaps you should create an issue dedicated to that

closed 2016-12-15 by chick

shunshou, 2016-12-15:
Would highly appreciate if there was Wiki documentation for how to do this, rather than having to comb through potentially incomplete/not-the-"one true way"-to-do-things code from other projects.