chipsalliance/chisel #268: Prepending JAVA_OPTS=-Xss256m does not fix StackOverflowError

matejaputic, 2016-08-28:
Following the tutorial, when installing FIRRTL, running:
`JAVA_OPTS=-Xss256m make build-scala`
gives the error:
`[error] (compile:compileIncremental) java.lang.StackOverflowError`

The full build output is at: https://gist.github.com/matejaputic/422bb807d1697139e3836a331c96f1b9

The output of `java -version` gives:
```
java version "1.8.0_101"
Java(TM) SE Runtime Environment (build 1.8.0_101-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.101-b13, mixed mode)
```

aswaterman, 2016-08-29:
What amount of stack size suffices for you?

matejaputic, 2016-08-30:
I'm not sure, I'm trying to install FIRRTL as a dependency of chisel3. How can I find out how much stack space is needed?

Step 2 of the [installation instructions for chisel3](https://github.com/ucb-bar/chisel3) is to install FIRRTL. The instructions state:
> If [`make build-scala`] fails with an infinite recursion / stack overflow exception, this is a known bug with JDK 8. You can either increase the stack size by invoking: `JAVA_OPTS=-Xss256m make build-scala` ...

I have tried that and it still results in the above error. I have also tried setting the stack size to 512m, which resulted in the same error.

Also, the message `Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0` appears at the beginning of the error log. It may be that this option is being set by the sbt install script, and is ignored by the JVM. However, I would expect to still be able to  increase stack space through a command line option, but what I have tried has not worked.

I am new to JVMs and I would appreciate your help, thanks.

aswaterman, 2016-08-30:
I'd just find out empirically, by increasing it by factors of 2 until it succeeds.  Hope you have lots of memory!

The MaxPermSize thing isn't an issue here.

matejaputic, 2016-09-01:
I have 32 GB of memory on this machine, and I have tried invoking this command with up to 16 GB of stack space. I think the problem is that my command line option is getting ignored.

"I have heard" that with Java 8, there is a global config that sets stack space, and that might be overriding my command line option.

How can I be sure that the option is not being ignored. Is there a more verbose mode with which I can run the compilation command?

kammoh, 2016-09-01:
I don't think the option is global in Java8 as I'm using the same JAVA_OPTS
both on oracle and Azul/OpenJDK. Could u check your sbt invitation script?
what OS are you running on?

kammoh, 2016-09-01:
sorry, that should've  been "invocation". Tend to make lots of typos when
replying on the phone :/

matejaputic, 2016-09-06:
I believe [this](https://github.com/ucb-bar/firrtl/blob/master/build.sbt) is the invocation script that comes with FIRRTL, and I haven't made any changes.

And from the looks of the [Makefile](https://github.com/ucb-bar/firrtl/blob/master/Makefile), it seems the `build-scala` target is just running the `firrtl.jar` file.

Is there an environment variable I could set, maybe, to increase the verbosity level or a logfile that might have a clue as to the variables that the JVM is configured with?

I am on Ubuntu 14.04.5 LTS

ducky64, 2017-02-06:
Is this still a problem? (if so, reopen it?)

closed 2017-02-06 by ducky64