chipsalliance/chisel #269: Deprecate either asUInt or asBits

colinschmidt, 2016-08-29:
Right now they do they exact same thing. I don't see the purpose for this, and it might make the user think they actually do different things. (I checked the source code to check that they were equivalent)

For what its worth, rocket-chip never uses asBits...

aswaterman, 2016-08-29:
asBits is nearly useless in practice, as it does the same thing as asUInt but returns a less concrete type.  I'd be fine with nuking it.  It wasn't in chisel2, so there's little compatibility concern.

sdtwigg, 2016-08-29:
Definitely +1 on deprecating asBits.

It had a nice symmetry with fromBits (asBits and fromBits were nominally
inverses) but we can probably introduce an alias for fromBits that restores
this.

ducky64, 2016-11-21:
Looks like this has been implemented, asBits is deprecated.

closed 2016-11-21 by ducky64