chipsalliance/chisel #271: Chisel compiler not catching value-less UInt declaration

zhemao, 2016-08-31:
This is probably easier to explain with an example. If you accidentally did the following

    val foo = Reg(init = UInt(width = 8))

Chisel will happily pass it through and then FIRRTL will give an error message like

    Exception in thread "main" firrtl.passes.CheckChirrtl$UndeclaredReferenceException$1: : [module DmaTestDriver] Reference T_653 is not declared.

Not sure if this is a Chisel or FIRRTL bug. But somewhere along the line, it should detect that you are using a UInt that isn't given a value and print out the proper Scala line number.

ucbjrl, 2016-08-31:
This is caught by Stephen's connection checks in the gsdt branch to be discussed in today's Chisel meeting.

ben-k, 2016-08-31:
This is a duplicate of #230 - perhaps close one or the other?

closed 2016-08-31 by zhemao