chipsalliance/chisel #276: ScalaDoc Issues

jackkoenig, 2016-09-08:
So I've started trying to use the ScalaDoc rather than just searching the code to help find issues. I'm creating this issue to help track the ones that I find. I should point out that I generate the ScalaDoc with `sbt doc` so if I need to do this differently, please let me know.

1. chiselFrontend stuff doesn't show up in the ScalaDoc -- this is a big problem given that things such as `Reg` and `Mem` live in chiselFrontend

mwachs5, 2016-09-09:
This has been driving me crazy as well. I was looking into unidoc (https://github.com/sbt/sbt-unidoc) as it seems like what we'd need, but my Scala plugin foo was lacking.

ucbjrl, 2016-10-06:
Closed by #301

closed 2016-10-06 by ucbjrl