chipsalliance/chisel #282: Flip doesn't work for Black Box IOs with Clocks

mwachs5, 2016-09-19:
This is perhaps related to #249 

If I create an IO Bundle with a Clock type, and then create a BlackBox with that IO.flip(), it doesn't get flipped in the black box:

Chisel Input:
```

class FooIO extends Bundle {
  val fooData = Bool(INPUT)
  val fooClk = Clock(INPUT)
}

class MyFooBlackBox extends BlackBox {
  val io = new FooIO().flip()
}
```

Creates FIRRTL Output:
`
extmodule MyFooBlackBox :
  input fooClk : Clock
  output fooData : UInt<1>
`

This causes problems later when Chisel tries to connect to the blackbox.

sdtwigg, 2017-02-03:
I want to say this is a manifestation of the direction bugs discussed in #440

ducky64, 2017-02-06:
Will try to fix with #440

ducky64, 2017-06-19:
Ok, looked into this and the problem is that the top-level `io` object doesn't actually get emitted for a BlackBox, instead its members become the top-level IO of a BlackBox and anything that applied to the `io` container itself is dropped.

This particular issue can be handled through special cases (like passing flipped, or even the user direction into `DefBlackbox` and having things emit accordingly), but if we start generating FIRRTL bulk-connects, then this splitting becomes a major issue since the `io` object *no longer exists*.

PR #655 (2017-07-28): Black box top-level IO fix
Resolves #282 in a hack-patchy way (direction of the top-level io object added to the DefBlackBox, since the io object itself is dropped during BlackBox generateComponent).

Also adds a test for a similar issue with regular Module direction, which may have been broken for entirely different reasons.
closed 2017-07-28 by azidar