chipsalliance/chisel #284: Illegal Argument: can't Mux between "heterogeneous types" of the same Bundle

ccelio, 2016-09-20:
Chisel3 checks that each element in a Mux is BOTH the same type AND the same width for each sub-field (which is different from Chisel2).

I think that's too strict. I believe Chisel3 should allow muxing between elements that are the same type, but each sub-field may be different widths (and the result should be the max size).

**The error message:**
````
Caused by: java.lang.IllegalArgumentException: requirement failed: can't Mux between heterogeneous types class boom.TageTableResp and class boom.TageTableResp
	at scala.Predef$.require(Predef.scala:219)
	at chisel3.core.Mux$.doAggregateMux(Bits.scala:749)
	at chisel3.core.Mux$.do_apply(Bits.scala:732)
	at chisel3.core.SeqUtils$.do_priorityMux(SeqUtils.scala:39)
	at chisel3.util.PriorityMux$.apply(Mux.scala:31)
	at chisel3.util.PriorityMux$.apply(Mux.scala:32)
	at boom.TageBrPredictor.<init>(tage.scala:217)
````

---------------
**Motivation:**

I'm currently porting my TAGE branch predictor to Chisel3. TAGE is composed of many tables, each table has a different number of elements, tag sizes, history sizes, and more. 

Thus, each TageTableResp will have elements of different sizes:

````scala
 class TageTableResp(fetch_width: Int, history_length: Int, index_length: Int, tag_sz: Int)(implicit p: Parameters)
    extends BoomBundle()(p)                                                                                        
 {                                                                                                                 
    val takens  = UInt(width = fetch_width)  // the actual prediction                                              
    val index   = UInt(width = index_length) // the index of the prediction                                        
    val tag     = UInt(width = tag_sz)       // the tag we computed for the prediction                                 
                                                                                                                   
    override def cloneType: this.type = new TageTableResp(fetch_width, history_length, index_length, tag_sz).asInst
 anceOf[this.type]                                                                                                 
 }
````

Each table provides its own prediction and I priorityMux the prediction from the largest table asserting a valid prediction.

````scala
187    val tables = for (i <- 0 until num_tables) yield                                           
188    {                                                                                          
189       val table = Module(new TageTable(                                                       
190          fetch_width    = fetch_width,                                                        
191          id             = i,                                                                  
192          num_entries    = table_sizes(i),                                                     
193          history_length = history_lengths(i),                                                 
194          tag_sz         = tag_sizes(i),                                                       
195          counter_sz     = counter_sz,                                                         
196          ubit_sz        = ubit_sz))                                                           
197       table.io.InitializeIo()                                                                 
198                                                                                               
199       // send prediction request                                                              
200       table.io.if_req_pc := io.req_pc                                                         
201       table.io.if_req_history := this.ghistory                                                
202                                                                                               
203       // update during mispredict                                                             
204       table.io.bp2_update_history <> io.hist_update_spec                                      
205                                                                                               
206       // check that the user ordered his TAGE tables properly                                 
207       if (i > 0) require(history_lengths(i) > history_lengths(i-1))                           
208                                                                                               
209       table                                                                                   
210    }                                                                                          
211                                                                                               
212    // get prediction (priority to last table)                                                 
213    val valids = tables.map{ _.io.bp2_resp.valid }                                             
214    val predictions = tables.map{ _.io.bp2_resp.bits }                                         
215    tables.map{ _.io.bp2_resp.ready := io.resp.ready }                                         
216    val best_prediction_valid = valids.reduce(_|_)                                             
217    val best_prediction_bits = PriorityMux(valids.reverse, predictions.reverse) // <<--- failing line!        
218                                                                                               
````

aswaterman, 2016-09-20:
It's not an unreasonable request, but the restriction stems from the difficulty of generating FIRRTL code for this case, and the axiom that Vec(a, b)(sel) should be the same as Mux(sel, b, a), and Vec must be heterogeneous.

ccelio, 2016-09-20:
I see the issue. So what's the solution for my code? Break out and priority-mux each element by hand?

aswaterman, 2016-09-20:
Hmm, can you make the offending field have inferred width instead of fixed, or use the same maximal width everywhere?  Otherwise, I've got nothin'.

ccelio, 2016-09-20:
Okay yuck, so my problem is larger than just the PriorityMuxing. At least that I can break apart to its constituent pieces and give each element its own mux. You're right, this goes into the Vec of Bundles with heterogenous widths issue.  :(

This is *really* useful for interfacing with the tables via dynamic indexes:

        val tables_io = Vec(tables.map(_.io))

I'll have to spend some time to see if I can really supply the maximal width. I'm afraid I'm going to have to track two widths for each element... the actual width, and the max width to make Chisel happy.

ccelio, 2016-09-21:
Okay, here's my workaround, where I provide the maximal width everywhere. It's... ugly. And opens me up to errors if I fail to handle cutting the width down somewhere. :(

https://github.com/ucb-bar/riscv-boom/commit/983e0f567dddafe42938ec8d353e6bce5690e22e

aswaterman, 2016-09-21:
yeah, that's pretty bad, sorry there's not a better solution at the moment.

why do you need to manually cut them down if the MSBs are zero?  are they concatenated?

ccelio, 2016-09-21:
Previously I could rely on width inference and the I/Os would cut things down properly. Now, I have to worry about out-of-bounds accesses to SRAMs/memories/Vecs, so it's worth being overly conservative.

As an example, `ghistory` is some length set by the machine and "cut down to size" for each TageTable's hashing function, which is used to index the prediction table. This hash-index is then passed out of the predictor, stored by the pipeline, and returned to the TageTable during commit.

grebe, 2016-09-22:
I'd just like to add on that this is causing me some issues as well. In my setting, I have a bunch of constants that are organized into Vec ( Vec ( SInts ) ). Creating Vec of Vec's triggers the same check as mux'ing (I believe for the very reasonable reasons mentioned by @aswaterman in the first reply).

Unfortunately, it is possible to choose literals that have different widths such that this check will fail. Even more unfortunately, it is possible to choose literals such that the check won't fail- maybe you get "lucky" when building a small version to get things working, only to find out that the real version won't work.

I suppose choosing some uniform big-enough width for every literal works, but being frugal with widths is kind of the whole point of the DSP stuff I'm working on. If I were muxing, I'd just mux elementwise, but I'm not muxing, I'm just building a Vec of Vecs. Any thoughts on what I should do?

sdtwigg, 2016-09-22:
If you are only doing static indexing (based off scala.Int, etc.) then using scala collections (like Vector, List, etc.) would work fine. Otherwise, if you need dynamic indexing you have to use a Vec and, since this dynamic indexing is effectively muxing, you need to have everything be sized homogeneously.

If you don't want to do the max-width determination yourself, making a Vec(Vec(UInt())) should work and then assigning it to it. The width inference system should figure out what size it needs based on what gets connected to it.

ducky64, 2017-02-06:
We can't solve this with Mux, since it must clone one of the input types, and you could get into a situation where one type has one width larger, and the other has another field larger, where the output would need to be a mix of both types.

You could write your own HeterogeneousBundleMux or something, where you know the Bundle constructor argument and could construct a Bundle that has fields with the largest widths.

closed 2017-02-06 by ducky64