chipsalliance/chisel #287: Chisel3 cannot create an 'all input' version of a Bundle

terpstra, 2016-09-21:
val a = DecoupledIO(UInt(width = 3))
val b = a.asInput
val c = a.asOutput
val d = a.flip

In this case, a==c and b==d. All four bundles have ready going in the opposite direction to valid.

If you want to write a Monitor for QA purposes, there is no way to take a bundle and force ALL the fields to be in the same direction. In other words, you can never write a Module which passively snoops on the traffic between two other components without unbundling the entire type.

@hcook Keeping you in the loop.

sdtwigg, 2016-09-21:
This should be fixed by my pull request since it rewrites how directions are propagated through the direction setting commands. (I haven't verified if other people's changes to try to preserve aggregate-level directions in FIRRTL broke anything.)

That said, I think the bug has to deal with the fact that, when overrideDirection is called, the flip adjustment only happens to the top-level (DecoupledIO). So, in the case of b and c, the directions of all nodes are set but not the flips of the leaves are not touched. This leads to consistency issues between dir and isFlip. Amusingly, the chisel core only seems to examine dir for root references when writing ports. isFlip is what is used in the Bundle type writings (and oddly, also as part of writing the port direction).

My proposed fix is, for asInput and asOutput, to blank isFlipVar for all nodes in the Data tree. Also, amend https://github.com/ucb-bar/chisel3/blob/a2cb95bfe9e9c30b73284e97048fa0187ab0ee1d/chiselFrontend/src/main/scala/chisel3/core/Module.scala#L101 to just look at direction and not the isFlip. Similar for the BlackBox port emission.

PS: Note that, calling asInput and asOutput on bi-directioned Bundles can lead to accidental surprises in other places. For example, stuff like https://github.com/ucb-bar/chisel3/blame/dda64c1dee16b5da15ac690bd3cd6759c3d5c032/src/main/scala/chisel3/util/Decoupled.scala#L74 will not work if you pass it a variant with the directions corrupted. (Also, that code was the only time a non-root dir was examined.)

PPS: You didn't end up testing this case; however, the direction setters are vulnerable to the standing cloneType issue where, for example, a.flip.flip is the same as a.flip. This is because all the direction setters first call cloneType and that wipes away all direction mutations (since the standard cloneType 'formula' does not tell the user to do direction updates).

ducky64, 2017-02-06:
Superseded by #440.
Input(...) doesn't do what we want it to do right now, but we're going to fix that...

closed 2017-02-06 by ducky64