chipsalliance/chisel #292: How to use multiclock in chisel3?

sashimi-yzh, 2016-09-23:
I want to use multiclock in the rocket-chip project as an FPGA design, with high frequency core and low frequency uncore. I change the `BuildCoreplex` and `BuildExampleTop` as following
```
    case BuildCoreplex => (c: CoreplexConfig, p: Parameters) =>
      uncore.tilelink2.LazyModule(new MultiClockCoreplex(c)(p)).module
    case BuildExampleTop => (p: Parameters) =>
      uncore.tilelink2.LazyModule(new ExampleMultiClockTop(p))
```
But this does not generate another clock input in the top module. So I try to add a new clock in the `ExampleMultiClockTopModule` class in `rocket-chip/src/main/scala/rocketchip/ExampleTop.scala`
```
val coreclk = new Clock(INPUT)
  multiClockCoreplexIO.tcrs foreach { tcr =>
    tcr.clock := coreclk
    tcr.reset := reset
  }
```
However, the firrtl compiler throws the following exception
```
Exception in thread "main" firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[ExampleTop.scala 67:15]: [module ExampleMultiClockTop] Reference coreclk is not declared.
```
It seems that I add the new clock in a wrong way. But the `Multiple Clock Domains` section in `Short Users Guide to Chisel` in the Wiki page has not been updated. I do not know how to add a new clock correctly.

mwachs5, 2016-09-23:
What if you try (I haven't tested this myself):

```

val coreclk = Wire(new Clock(INPUT)
...
```

mwachs5, 2016-09-23:
Actually, that's probably not right... If you want the coreclk as an input, declare it in the io:

```
val io = new Bundle {
...
val coreclk = Clock(INPUT)
...
}

tcr.clock := io.coreclk
```

sashimi-yzh, 2016-09-23:
Thank you for the help! I declare the new clock in the `BaseTopBundle` class, and let  io.coreclk connect to trc.clock. This gives me the new clock input in the top module.

closed 2017-02-06 by ducky64