chipsalliance/chisel #294: Standardize the One True Way for defining literals

ducky64, 2016-09-24:
With a plan for merging #265 on the roadmap, there are now quite a few (inconsistent!) ways to specify literals:
- UInt(value, width)
- UInt.Lit(value), UInt.Lit(value, width)
- value.asUInt(), value.asUInt(width)
- value.U

And several ways to construct value-less fields:
- UInt(width)
- UInt.width(width)

(and I'm probably missing a few, please add them...)

Since we'd like to break as much of the API as we need to before chisel3 (to minimize continued breakage in the future), let's try to standardize on one. The rest should be deprecated and/or moved into compatibility packages (so existing chisel2 compatibility code continues to work), and users can continue defining their own in their own codebases if they like, but officially, chisel3 should have just one.

I'm of the opinion that UInt(width) and UInt(value, width) should be mutually exclusive, because the first parameter isn't consistent in both cases, and they also do very different things.

Whether we keep UInt.Lit(value, [width]) or value.asUInt([width]) or value.U is a cosmetic and style question. .asUInt is structurally similar to Scala's toInt() methods, and the distinction between .U and .asUInt is just verbosity (though there's a good argument of favoring readability over code conciseness, before your code turns into symbols and alphabet soup)

Thoughts?

aswaterman, 2016-09-24:
IMO, 1.U is also more readable than 1.asUInt; the extra chars are clutter.
This is a DSL. The distinction between Int and UInt should not be so
strong. It would be ridiculous if it took fewer chars to express a literal
in the IR than in the language.

aswaterman, 2016-09-24:
BTW, I don't think the PR should be merged til the exact syntax is
resolved. This is about the API, after all!

ducky64, 2016-09-24:
That's true, but the other argument is that this could be a much longer debate (and arguably an issue that wasn't introduced by the PR, though it certainly contributed to the addition of styles). Note that there are some other issues still being worked out with the PR, so it's entirely possible this debate could be resolved before the PR is ready, and the results integrated.

More people should chime in: @jackkoenig @sdtwigg @chick @ucbjrl (probably many more I missed)
Including the DSP people, and whether their DSP DSL can be stylistically consistent: @grebe @shunshou

shunshou, 2016-09-24:
Hopefully @grebe will respond, but I think he was saying that there was some bug with Vec of Vec of Lits? Something about Lits just taking on the minimum width they needed to take on, but due to not supporting non-homogeneous Vecs [I assume for some ROM thing?], there was some error thrown...

And his intended work around was something like first declare UInts of unknown widths and then assign them to Lits... ???

---

Regardless, I think it's useful to be able to specify a Lit where it auto calculates the width, and, if there's some chance that code might break (as in @grebe's case and possibly others, although I can't think of any atm), specify a Lit where you can give it a width that it would pad to. 

1.U and -1.S seem cute? I haven't played around with Chisel3DSP front end yet (next week!), but I would assume you wouldn't be able to do something like -1.333.F for Fixed pt vals? (can you have two "."'s? and you'd still need to specify # of fractional bits...)

I believe @chick @azidar and @grebe have picked some way to represent Lit's... I'm personally ok with whatever as long as I get to use them in the ways I'd like to use them...

grebe, 2016-09-24:
I like 1.U. I forget if you need the dot. @shunshou is right to bring up -1.333.F, but I think that it works (or if it doesn't, then (-1.333).F does, and if you don't need the dot maybe -1.333 F will work). I think the real issue is that it looks a bit funny. I'm positive that @chick showed me some examples that made me feel good but I'm forgetting exactly what they are. I don't think we've thought a lot about what literals should look like for Complex numbers, but I don't see a problem with Complex(real.U, imag.U) or whatever.

The Vec of Vec issue I think is orthogonal to how you actually specify literals, as long as there is a way to specify the width of a literal (which I feel is extremely necessary).

I also like UInt.Lit and getting rid of UInt(value), but of course the problem is fixing all the code that uses UInt(value). I don't think there's anything particularly special about DSP here.

ducky64, 2016-09-27:
@grebe the main idea here is to specify the One True Way for chisel3. The ways to want to remove can be deprecated for a few versions before being removed and added to chisel2 compatibility mode, which avoids breaking anyone's code immediately. RTL which makes extensive use of deprecated constructs and which really don't want to migrate can define their own compatibility layers past deprecation.

Anyways, what I'm getting here is that:
We want to keep 1.U over 1.asUInt. Perhaps also add 1.U(width) for consistency?
Kill UInt(value), UInt(value, width) with fire. Probably keep UInt(width) (it's sufficiently different from a literal constructor)?
Not sure what to do about UInt.Lit. Some like it, though it would be duplicated with 1.U now.

sdtwigg, 2016-09-27:
The main temptation behind (at least retaining) asUInt is that I think it
reads better in more cases than just .U.

When just instantiating from actual scala constants without width
specifications like with 1.U and 1.asUInt, either looks fine.

When instantiating from scala identifiers like maxElements.U and
maxElements.asUInt, just U can 'get lost' or be more unclear since you
don't have the context of seeing a literal number to jump off from. This
also applies to writing expressions like (0 until 8) map (_.U) versus (0
until 8) map (_.asUInt)

When specifying a width in both cases, asUInt(width) provides a nice
symmetry with the UInt(width) for constructing UInt-types. e.g. 1.asUInt(2)
would be something we can definitely put into a Reg(UInt(2)) and matches
visually.

Even if you make it package private, you should still have Lit or some
variant in the constructor so that the actual code for constructing a
literal UInt is 'close' to the UInt class definition. The stuff in the
package object implicits then is just using a simple function call to
delegate to the more complex code.

ducky64, 2016-09-27:
So... keep both .U and .asUInt (with the former intended for constructing basic numerical literals, and the latter for readability when constructing literals programmatically - perhaps as part of style guidelines)?
And make .U and .asUInt just call into a UInt.Lit (chisel3-private), which is the internal way to construct literals?

So UInt(value, width), UInt(value), UInt.width(width), and UInt.Lit (all variants) would be deprecated from the public API?

Do we want to add .U(width)? Probably not, since it's not particularly readable, especially if we're keeping .asUInt(width)?

ccelio, 2016-09-27:
UInt(width) is awful, it goes against tons of existing Chisel code (and tutorials and docs) and it's not intuitive.  I'm not sure why the hate against UInt(value) and UInt(value, width), other than we needed a better ability to throw errors/warnings when people did things like Reg(UInt(4)) instead of Reg(UInt(width=4)) or Reg(init=UInt(4)).

ducky64, 2016-09-27:
What should the API be for a constructing a value-less UInt of specified width then?

(my reason for the hate against UInt(value) is because of the potential confusion you stated, I prefer an API that makes things unambiguous, especially for people who haven't had years to learn the quirks)

ccelio, 2016-09-27:
I've always been quite content with UInt(width=x); it's very clear.

ducky64, 2016-09-27:
As of current Scala, macros break with named and default arguments. I don't think we've needed macros with UInt yet, but this may limit options in the future.

Finding the constructor for UInt(width=x) also isn't very clear, behind the scenes there's a default dummy first parameter that disambiguate it from the UInt(x) case.

SebastianBoe, 2016-10-16:
Maybe I am missing something, but I was surprised to find out that
assert((UInt(1) + UInt(1)) === UInt(2))
started failing in Chisel3. Could not find anything in migration guides
about this.

I hope UInt(x) is either removed due to it's ambiguity or kept to still
mean UInt(value).

aswaterman, 2016-10-16:
It is confusing, but this is actually a bug fix. In both chisel versions, + is modular addition. Since UInt(1) is one bit wide, the result of adding UInt(1) to UInt(1) should be one bit wide, i.e. UInt(0).

Chisel2 performed the modular addition of one or both inputs was not a literal, but incorrectly performed exact addition if both were literals.

To get the behavior you desire, there's a new operator, +&, which performs exact addition.

aswaterman, 2016-10-16:
Oops, I replied before seeing your edit.

SebastianBoe, 2016-10-16:
Sorry about the noise. Great to see +& entering the language, much cleaner than Cat'ing with 0.

shunshou, 2016-10-16:
Can I still beg that eventually + not be modular addition and instead use a separate +% for modular addition?

aswaterman, 2016-10-18:
Since it has been like that for 6 years, the backwards incompatibility is likely to cause more confusion than it solves.

ducky64, 2016-10-24:
Bump bump bumpity bump. We should really (at least!) resolve the One True Literal Constructor.
(and we could still keep all the legacy code working without rewriting through compatibility shims while presenting only one clean way in new code)

ducky64, 2016-11-16:
We discussed this at the Chisel meeting today, and the proposal we came up with for chisel3 going forward (as in, no removals / deprecations to Chisel compatibility package) is:
- Constant literal specifier: `2.U` and `myInt.asUInt`.
 - Deprecate ~~`UInt(value, width)`~~, ~~`UInt(value)`~~, ~~`UInt.Lit(...)`~~.
- Width specifier: `UInt(width)`, where `width` must be of Width type.
  - Deprecate ~~`UInt.width(width)`~~ and ~~`UInt(width=x:Int)`~~
  - Previous code written with ~~`UInt(value)`~~ will type error and be "safe". Because of the different type signatures, this could also be backported to Chisel2.
  - We will add a shorthand specifier to construct Width from a Int, like `4.W`. Exact semantics up for debate.
  - This makes our version somewhat similar to certain forks that have a ~~`UInt(width: Int)`~~ constructor, where width is just an Int. Implicit conversions (_won't_ be part of standard Chisel) can bridge the gap.
  - The `.U` and `.asUInt` constructor will be consistent with the `UInt` constructor, for example `2.U(4.W)` would create a 4-bit wide literal with value 2.
  - Future: the width field in the constructor would be generalized to handle ranges. The ranging PR is coming soon, but the whole constructor would look something like `UInt(range"[0, 32)")` (a UInt from 0 inclusive to 32 exclusive)
  - This makes the difference between Chisel data types, Scala numbers, and widths explicit and exposed in the API.

Thoughts? @aswaterman @shunshou @sdtwigg @azidar @jackkoenig

shunshou, 2016-11-16:
I'm all for being explicit. Hopefully everyone agrees on this as a convention and I can start using it, instead of having to recode everything for the n-th time. :p

aswaterman, 2016-11-16:
Seems reasonable.

Re: ranges: it's important that widths and ranges are kept entirely separate, rather than implementing widths in terms of ranges.  A width of N is not the same thing as a range [0,2^N-1], because the ranges allow optimizations that would not be allowed for widths, changing the semantics.  This goes for both Chisel and Firrtl.

shunshou, 2016-11-16:
So how do you propose two numbers (one specified w/ worst case width; the other specified with a possible range) be added? Seems to make sense to add range + width converted to range in that case. 

Generally speaking, I probably wouldn't be mixing and matching too often, if at all, but...

shunshou, 2016-11-16:
Or is it better then to have some kind of RangedUInt that gets propagated throughout if you want to be able to differentiate between the two easily?

aswaterman, 2016-11-16:
Mixing and matching them is fine. If you add a ranged thing with a
widthed thing, you get a ranged thing.

If you add a widthed thing with a widthed thing, you should get a
worst-case widthed thing, not a slightly-more-optimistic ranged thing.

UInt literals must be widthed by default, not ranged, to preserve current
behavior.

The width inference problem rcan't be expressed as a range inference
problem, because it is incorrectly optimistic.

I'm saying that if you add two width'd things, you should get a worst-case
width'd thing, not a ranged thing. And the width inferencer can't be

azidar, 2016-11-16:
I believe it is possible, if both width and range information is kept for the inferencer, to do range optimization on width-declared wires only when the effects of optimization aren't visible. For example, if you always take the lowest bit from an addition, you can certainly back-propagate that constraint to trim widths. This like head, bits, tail, etc. should dictate these requirements.

shunshou, 2016-11-16:
I'd say that you'd likely get the most optimization when treating Lits (with unspecified widths) as having a range of [itself,itself]. But that probably doesn't matter if you expect to be adding a Lit to a ranged thing.

shunshou, 2016-11-16:
Re: lowest bit from an addition, you then have to separate out +% and + behavior? Or are you assuming +% always is kind of "widthed" vs. + is "ranged" = would never overflow?

aswaterman, 2016-11-16:
As long as it has zero effect on the widths of width-inferred things vs.
current algorithm, that's fine. The extra info can then be used to
optimize ranged things.

aswaterman, 2016-11-16:
@angie that would be backwards incompatible. Current code can correctly assume that 1.U+1.U+1.U has a width of 3.

aswaterman, 2016-11-16:
Sorry I meant +& not +

azidar, 2016-11-16:
Firrtl only has one add operator, so if ranged UInts want to use that behavior, they need to use %+.

aswaterman, 2016-11-16:
I'm talking about Chisel operators. Anyway, my point is, make sure Chisel's current behavior is exactly preserved.  Range inference is a new feature and should have absolutely no effect on code that doesn't use it.

grebe, 2016-11-16:
I believe that @azidar and @chick and I had a discussion some time ago about how to do ranges with the same width-inference behavior. The idea was that wherever range inference could trim something, it would produce a UInt/SInt with a width the same as the normal width inference, but those bits that can be trimmed would be replaced with constant 0s/sign extension.

aswaterman, 2016-11-16:
Probably possible. I propose large designs passing a formal equivalence check should be required before merging that. It's a little scary.

grebe, 2016-11-16:
I agree. I think there should always be a setting so that range inference reverts to "conservative" width inference

shunshou, 2016-11-16:
You mean constant 0's / sign extension [up to full width] that you then leave the synth tools to take care of presumably b/c they're good at that? I guess I buy that. 

I mean, i thought in that case, you might as well make range inference an optional pass that you can run on top of width inference if you so desired...

aswaterman, 2016-11-16:
The separate-pass approach seems more prudent to me.

shunshou, 2016-11-16:
In that case, you'd have to probably be a bit more choose-y on modules to perform the pass on [i.e. don't touch rocket-chip related modules b/c you guys like your bitwidths but get optimization-happy with DSP] -- and in that case, is there an easy way to group modules you don't want to touch without having to go through things one by one if, for example, rocket-chip is actually top level?

aswaterman, 2016-11-16:
Module annotation?

shunshou, 2016-11-17:
Sure. Just annotate which modules are "safe" for range inference, I guess? It's just weird to me when you're hierarchically building something from a bunch of sub projects [rocket, DSP block 1-100, etc.], and you have to "manually" annotate which is safe. But then again, you probably shouldn't be annotating as the systems guy, and it should be each individual block designer's job to annotate whether his/her block is range inferrable. 

My problem right now is (well, at least in the summer), when you use rocket-chip, you're kind of locked into its ecosystem and you have to abide by its rules unless you compile your stuff separately and black box it.

azidar, 2016-11-17:
I mean, ideally the optimization should never affect any behavior, regardless of whether you declare a wire with a range or a fixed width. So in the long term (after verification), this optimization should be fine to run on rocket-chip. If in the interim you want to selectively run it, that's fine, but we should make this goal clear.

shunshou, 2016-11-17:
In some contexts, it makes sense to think of stuff with only widths... so ranges shouldn't apply to everything. I also really don't like that Bits is equivalent to UInt. I know you've discussed this with Paul a bit, but what does bitwise and, or, etc. mean in the context of a UInt? IMO, you should only be allowed to do that if you've explicitly casted the UInt to be Bits. 

Even if you could theoretically consider ignoring stuff in a range optimizer, I think it's important to separate out numbers that have range from bits that you want to do bit manipulations on [which is more of a processor thing?]...

I thought part of the point of Chisel was to be type-safe, but if you confuse Bits/UInts, you're not really abiding by the philosophy, IMO.

aswaterman, 2016-11-17:
UInts are defined to use standard binary encoding and SInts are defined to use two's complement, so the bitwise operators are pretty clearly defined.  Also, Verilog, C, Java, Scala, etc. define bitwise operators on integers for the same reason, so it seems least surprising to permit them.

Bits == UInt is a wart from chisel1.  The old Bits companion object should be deprecated in chisel3.  It's not actually possible to construct a concrete Bits, anyway, since it is an abstract class.

aswaterman, 2016-11-17:
In any case, range inference works fine for bitwise operators, too.  The rules are just more complicated than for arithmetic.

azidar, 2016-11-17:
@shunshou, are you saying that you want the distinction between UInts and "Bits" because you don't want ranged optimizations to happen on things you intend to be "Bits"? If that is your worry, it shouldn't be. Again, any ranged optimization will not affect the circuit. Thus, any use of a UInt in a "Bit" context will have its exact behavior preserved.

> I thought part of the point of Chisel was to be type-safe, but if you confuse Bits/UInts, you're not really abiding by the philosophy, IMO.

Its only "not abiding by that philosophy" if there is an observable difference between the two. Since there isn't, there is no need to add additional distinctions with no added benefit.

shunshou, 2016-11-17:
I just don't think the output of a bitwise operation should be allowed to have a range, unless you're doing something more general like selecting [Mux] between 2 numbers, even if ranging doesn't affect the circuit.

Like do you ever associate ranges with boolean algebra? Maybe someone who knows more should fill me in...

aswaterman, 2016-11-17:
It seems weird to prohibit, since it's just another operator.  As Adam points out, it should be invisible to the user, anyway, so why does it matter what optimizations the compiler is performing?

An example where this would actually show up is in floating-point rounding hardware.

shunshou, 2016-11-17:
Ok, guess it doesn't matter. As long as Chisel functionality keeps growing, I'll be content...

ducky64, 2016-11-21:
After almost two months, this is finally done!
(the discussion here also went off a tangent on ranging, that should probably be tracked in a separate issue if it's still an outstanding item)

closed 2016-11-21 by ducky64