chipsalliance/chisel #298: Issues with getFirrtlDirection in gsdt

jackkoenig, 2016-09-28:
Users reported the following code to be broken in gsdt:

```scala
val queue = Module(new Queue(UInt(width = 32), 8))
Irrevocable(queue.io.enq)
```

This is simply trying to cast the enq port of a Queue to IrrevocableIO so it can be connect to by IrrevocableIOs. It causes the following requirement from the apply method of Irrevocable to fail:

```scala
require(getFirrtlDirection(dec.bits.flatten) == INPUT, "Only safe to cast consumed Decoupled bits to Irrevocable.")
```

This is obviously absurd as enq.bits (being `dec` in the above require for "decoupled") is clearly an Input (side note, it is super confusing that enq in Queue is a DeqIO and deq is an EnqIO). Also note that deleting the requirement results in correct Firrtl code.

I tracked down the issue to the rather pedestrian observation that getFirrtlDirection is defined as returning  "firrtlDirection is the direction we report to firrtl.                                                                                                                                                  It maintains the user-specified value (as opposed to the "actual" or applied/propagated value)." This means the check is wrong because it doesn't take into account that this bundle may be a part of a flipped bundle, thus the failure above because enq is Flipped so the real direction on enq.bits is the opposite of what the require is seeing.

In my opinion, this calls into question the utility of getFirrtlDirection, or at least the fact that it returns a chisel3.core.Direction even though it's not actually even *intended* to give a Chisel Direction.

This code works if we replace the predicate of the require with:
```scala
dec.bits.flatten forall (_.dir == INPUT)
```
where we actually check each field (recurisvely) of the Bundle for its Chisel direction.

ucbjrl, 2016-09-28:
Excellent analysis @jackkoenig. `firrtlDirection` should only be used when emitting firrtl.

W/r/t the issue of enq and deq in Queue, the perspective is from the client of the Queue. Internally, the Queue object uses the flipped version of the interface used by the client.

ucbjrl, 2016-09-28:
Closed by #299

commit 2edfe895 (2016-09-28): Don't use firrtlDirection for direction checks - fix #298.
firrtlDirection should only be used for emitting firrtl. Any checks on the actual direction should use the bound Direction `dir`.
commit 12a65151 (2016-09-28): Merge pull request #299 from ucb-bar/gsdt-decoupleddirection298

Don't use firrtlDirection for direction checks - fix #298.
closed 2016-09-28 by ucbjrl