chipsalliance/chisel #300: Use Builder.error where possible

jackkoenig, 2016-09-28:
This is especially relevant to the gsdt branch

When possible (eg. a check returns Unit), we should call Builder.error instead of throwing an exception. This allows Chisel to continue and possibly pick more errors rather than immediately ending elaboration.

sdtwigg, 2016-09-29:
This is a bit tricky. For functions that don't actually return anything (like connections) Builder.error is fine. Although note, in some cases you need to print the whole stack trace for debugging and thus having multiple errors all printing their stack traces can get a bit spammy.

Functions that return something (particularly, cloneType) can't really just register an error because they don't have a valid thing to return. For example, if cloneType fails currently, it registers an error and then returns this (which is bound to other things). The code then starts doing modifications to 'this' (instead of a fresh object). This generally results in a multitude of even stranger errors occurring as a result of everything failing due to that constraint being violated.

sdtwigg, 2016-09-29:
Also, (and I forgot that stock Chisel still has this problem with dynamicContext), the error log is stored as part of the dynamicContext. In the case where the dynamicContext is not setup properly (is None), all errors are silently suppressed.

ucbjrl, 2016-10-04:
We could have a special Error class (subclass of Data) an object of which can be returned when a failing function must return something. This object could be immune from triggering other errors.

ducky64, 2017-02-06:
Resolution: where we can return something sane, we should. Otherwise, error on the spot so you don't get a pile of 50+ errors after it.

This issue is too general. If there are specific places where you think chisel should store an error and keep going, file those as separate issues.

closed 2017-02-06 by ducky64