chipsalliance/chisel #303: Better name propagation by macros, RFC

ducky64, 2016-09-29:
So, this has been a requested feature for a while (informally as well as formally, see #262). Basically, wire name propagation to FIRRTL right now only works for vals that are defined directly within a module (not inside a sub-scope, such as a if block, where it wouldn't be visible to reflection) and by explicit TransitName (which isn't widely used, and in my opinion has a somewhat clunky API and implementation).

## Basic Idea
The gist of this proposal is to use static annotations (a macro paradise feature) to rewrite each instance of
`val name = expr` into `val name = Namer(expr, name)`, where Namer is a passthrough function that sets the object's name field (if it has one, and does nothing otherwise).

Note that this is the same basic approach as used in @sdtwigg 's gama Chisel fork.

## Modules
The case for naming in Modules (and sub-blocks directly in Modules, like when or if blocks) is pretty simple, the transform above works. In cases of name aliasing, Chisel seems to be able to resolve it by adding a _1, _2, etc. One thing that might be nice is to have string interpolation into naming; for example:
```
for (i <- 1 to 10) {
  val myInt_i = i.asUInt
}
```
where i would be interpolated, so in FIRRTL you would get names like `myInt_1`, `myInt_2`, ..., `myInt_9`.
I'd like suggestions for a syntax for this though.

This would strictly be opt-in, users would need to annotate their modules with a static annotation, like `@module`:
```
@module
class myModule extends Module {
```

This also requires users to compile with the macro paradise plugin, though Chisel RTL that doesn't use the annotation would still continue to run fine without macro paradise.

## Functions
However, gets gets funner when dealing with propagating names to functions, since the usual pattern (I think, please correct me if I'm wrong) is:
```
val myQueue = Queue(blah)
```
where Queue is a function that instantiates a sub-module and might also contain some logic. But because the sub-module is instantiated in a different function, we would get the same generic name, q, that's defined in Queue(). That's obviously not helpful, and was the basis for the post-name hooks in TransitName. I'd like to replace that with macros too, hopefully making a syntactically consistent naming system that doesn't need changes in code.

_The rest of this section is more speculative and has not been backed up by a proof-of-concept implementation yet._

One possible way is to rewrite a function definition to take an implicit EnclosingName, and provide the implicit on val definitions:
```
object Queue {
  def apply(enq: ..., ...) {
```
into
```
object Queue {
  def apply(enq: ..., ...)(implicit enclosingName: EnclosingName) {
```
And all val invocations would be transformed into:
```
val name = {
  implicit val enclosingName: EnclosingName = EnclosingName(name)
  Namer(expr, name)
}
```

### Issues
However, one problem here would be vals in vals, for example, this untransformed code would likely result in a implicit resolution error:
```
val myUInt = {
  val dummy = UInt(0)
  dummy
}
```
The simple solution here would be to prevent recursively transforming expressions, so dummy wouldn't get named. Other ideas are welcome, too.

Another issue would be with chained apply, since the chained apply parameters would be confused with the implicit parameter list. This could be solved by the explicit-implicit pattern (as with SourceInfo macros). Macros may also provide an automated transform of the function definition, instead of the manual `do_*` functions used in SourceInfo.

### Cooperating with other implicits
With the gsdt features coming up soon, functions may also end up taking a CompileOptions implicit. SourceInfo already uses implicits - currently only in chiselFrontend, but it may be expanded into a user-facing API in the future allowing multiple levels of source locators (such as indicating a library entry point). We need some mechanism to allow all the implicits to play nicely with each other, and a bonus would be if they all used a similar underlying mechanism.

One possibility is to add these implicits to functions using StaticAnnotations into the function's normal parameter list:
```
defn myFunction(myArgs)
```
would be transformed by the StaticAnnotation into
```
defn myFunction(myArgs) = macro myImplicitTypeTransform
defn do_myFunction(myImpicit: MyImplicitType, myArgs)
```
where myImplicitTypeTransform would turn calls from
```
myFunction(myArgs)
```
into
```
do_myFunction(implicitly[MyImplicitType], myArgs)
```
Once again, this is speculative, but I _think_ it should work. StaticAnnotations should be resolved in order, and if the macro implementations are written appropriately, everything should chain correctly. I'm not 100% clear on what the full implications are, though I think it addresses some common sticking points like we saw with the implementation of SourceInfo.

## Code
See the development `nameprop` branch.

aswaterman, 2016-09-30:
Note that SourceInfo is exposed and is already used extensively in the TileLink2 implementation to provide better source locators in our own deeply nested library code.  It is great!

Re: Issues, I think not recursively transforming expressions is probably OK.  Better to be backwards compatible, and use better naming as a carrot for people to write code differently, than to use the stick and break the old code.

colinschmidt, 2016-09-30:
As a maintainer of some deeply nested code I agree with Andrew that I'd rather have a best effort while maintaining working code than a perfect naming with a bunch of broken code.

mwachs5, 2016-09-30:
I hope one thing that would be accomplished in this task would be to have module naming be predictable and unaffected by other things you may instantiate elsewhere in your design.

For example, say I  have 3 ParameterizedModule instantiated in some module of my design, using functions of course. Since they are all different internally, they get uniquified. The inferred instance names have the uniquified module name in it, so their instance names may be something like:

```
ParamterizedModule ParameterizedModule_1 (...)
ParameterizedModule_1 ParameterizedModule_1_1 (...)
ParamterizedModule_2 ParameterizedModule_2_1 (...)
```

Now I add, somewhere completely different in my design, another instance of ParameterizedModule, which happens to be the first one encountered so it gets the "ParameterizedModule" name. All the instance names above get changed as a result.

This sort of unpredictability of the hierarchy causes problems when writing backend constraints, etc.

ducky64, 2016-09-30:
So I don't think I can control auto-naming order, though perhaps extending the name interpolation mechanism suggested above to module names may help? If your modules are completely described by the parameters passed in, then a unique module name should prevent the autonaming stuff from messing things up at all?

As a side note, I think the current system automatically tries to generate instance names from module instances that are val definitions within the top level scope of the a module.

mwachs5, 2016-09-30:
I think the module name is even more important than the signal name. You're right that the name interpolation works for things instantiated in top level modules (it's great!). 

A simple thing would be for inferred names, use the non-uniqified *module* name to create the instance name? In the above example, the names could have been instead:

```
ParamterizedModule ParameterizedModule_1 (...)
ParameterizedModule_1 ParameterizedModule_2 (...)
ParamterizedModule_75 ParameterizedModule_3 (...)
```

I have no idea how all this works internally or how crazy this request is.

ducky64, 2016-09-30:
I'm not sure using the non-unique'd module name is helpful? I guess it would prevent instance names from changing with module names, but it feels like a hack and seems counterintuitive in other situations. val name propagation here should also remove the need for non-unique'd names

Also, for name interpolation, I mean splicing the variable value into the name. For example, if you had
```
class Queue__n_(val n: Int)
```
(of course, the syntax is up for debate, probably could make it either an annotation or a naming call within the module body)
and an instance
```
val myQueue = new Queue__n_(10)
```
the value of `n` would be spliced into the name of the module, so you would get
```
inst myQueue of Queue_10
```
Note that this is different than just normal name propagation, which currently exists using reflection. As far as I know, the current name propagation system doesn't have any kind of interpolation mechanism (and it really wouldn't need one, because it can't look inside loop bodies anyways)

sdtwigg, 2016-09-30:
The naming of module definitions is orthogonal to the proposal; however, #251 granted the ability to alter the desired Module name to fit your needs. In the case of parameterized Modules, you can just amend the string to incorporate the parameter name. It seems like, if backend uses desire stabler names for certain Modules, you would just want to ensure at least those specific Modules have overridden default naming.

mwachs5, 2016-09-30:
Yeah, at this particular moment my pain point is the instance names, not the uniquified module name. 

And what @sdtwigg is suggesting is what I'm doing right now, adding "suggestNames()" where I need them to maintain my sanity, and I'm fine with that solution for now.

sdtwigg, 2016-09-30:
In response to my own edit:
Do you think your pain would be lessened or increased by in:
https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/Module.scala#L156
changing m.name to m.desiredName ?

mwachs5, 2016-09-30:
I don't have a good enough grasp of the Chisel internals to understand the implication of that code snippet... if you give an example of the impact I'd be happy to comment,

sdtwigg, 2016-09-30:
In short, the change should immediately do what you asked for in
https://github.com/ucb-bar/chisel3/issues/303#issuecomment-250804593

In longer form,
Both name and desiredName refer to the module def name.
Module.name is a uniqified variant of Module.desiredName.
So, for a bunch of Modules subclassing Queue, all will have desiredName "Queue"
However, each variant will have a uniquified name (say Queue_1, Queue_2).

With m.name, if a module uses multiple Queues then the _# introduced as part of module def uniqification will be used to suggest the instance name (on top of another _# added to disambiguate between instances).
With m.desiredName, you just get one _# due to instance disambiguation.

Note that, with desiredName, you can get some other weirdness since all Queues (regardless of parameterization) will compete for the name Queue, so you may see:
Queue_2 Queue_1 {...} // Queue for UInt32
Queue_1 Queue_2 {...} // Queue for UInt256
However, you do get a much better guarantee that someone else adding/removing a Queue definition won't change your instance names. So, things may get reshuffled but you still get
Queue_5 Queue_1 {...} // Queue for UInt32
Queue_7 Queue_2 {...} // Queue for UInt256

mwachs5, 2016-09-30:
yes, I think that would be better and more expected. But it may be breaking someone else's problem.

shunshou, 2016-10-03:
Out of curiosity, would appending source location info as part of an instance name [in Verilog] be useful? It's more informative than _1, _2, etc. if something like Queue is used everywhere...

colinschmidt, 2016-10-03:
I just want to mention that this better naming would help with an issue I faced recently getting multiply instantiated modules working in chisel3. You can see the two PRs I opened on [rocket-chip](https://github.com/ucb-bar/rocket-chip/pull/375) and [hardfloat](https://github.com/ucb-bar/berkeley-hardfloat/pull/9)

ducky64, 2016-10-03:
@colinschmidt the proposal probably wouldn't directly fix what you had in rocket-chip, unless you rewrote it so everything named is a val, say:
```
def isAMO(cmd: UInt) = {
  val cmd3Wire = cmd(3)
  val mXaSwapWire = (cmd === M_XA_SWAP)
  val isAMOWire = cmd3Wire || mXaSwapWire
  isAMOWire
}
```
(inferring good names that are the result of logic may be a future feature, but that comes with its own set of issues)

As for hardfloat, isn't instance val naming already part of Chisel (as long as the instance vals are directly in the containing module, not within a sub-scope)? What FIRRTL was it producing before?

ducky64, 2016-10-03:
@shunshou What do you mean appending the source location? It seems like it would make a lot of names very long, like myWire_SomeScalaFile_49_4, as well as making names sensitive to their position in source Scala (which is highly likely to change)...

colinschmidt, 2016-10-03:
@ducky64 previously in hardfloat it would name them
```
inst RoundRawFNToRecFN_10_1 of RoundRawFNToRecFN @[RecFNToRecFN.scala 102:19]
```
and 
```
inst RoundRawFNToRecFN_22_1 of RoundRawFNToRecFN @[RecFNToRecFN.scala 102:19]
```
In the two instances of the outer module.

aswaterman, 2016-10-03:
@ducky64 I agree, putting source locators in names would reduce the stability of naming.

@colinschmidt I think I know the culprit.  Instead of

    // For Module instances we haven't named, suggest the name of the Module
    _ids foreach {
      case m: Module => m.suggestName(m.name)
      case _ =>
    }

we should be doing

    // For Module instances we haven't named, suggest the name of the Module
    _ids foreach {
      case m: Module => m.suggestName(m.desiredName)
      case _ =>
    }

The problem is that chisel is using the disambiguated module name to inform the instance name.  It should use the possibly-ambiguous model name instead.

shunshou, 2016-10-03:
Assuming some custom name isn't provided, there is still quite a bit of usefulness in somehow propagating source location (ok maybe file name + line # is enough) all the way through. I've been annoyed by having to stare at Verilog waveforms for intermediate nodes and having not a good sense of where they came from. It'd be nice if, when I don't enforce some kind of name and just let Chisel do its thing, I could refer back to the line it was created...

aswaterman, 2016-10-03:
Is there an annotation format that DVE understands?  That sounds like a more appropriate way to connote this information.

colinschmidt, 2016-10-03:
@aswaterman Unfortunately that doesn't solve the issue. changing it to desiredName removes one of the _# from the name
```
inst RoundRawFNToRecFN_11 of RoundRawFNToRecFN @[RecFNToRecFN.scala 102:19]
```
```
inst RoundRawFNToRecFN_13 of RoundRawFNToRecFN @[RecFNToRecFN.scala 102:19]
```
But they still end up distinct.

aswaterman, 2016-10-03:
Then it looks like there are two distinct issues at hand.  m.name was definitely part of the problem.

aswaterman, 2016-10-03:
FYI, what actually happened is the _first_ set of _# was removed, and the second set remained; but _1 no longer suffices because there's more collision.

ducky64, 2016-10-04:
Back to the proposal for propagating names into functions (using implicit variables), there's a possible edge case if hardware generator functions are passed into hardware generator functions.
For example, if we have
```
val myQueue = Queue(UInt(width=2))
```
where Queue is defined as 
```
def Queue(...) = {
  val q = Module(new Queue())
  q.deq.io
}
```
then with name propagation through functions, the FIRRTL would look something like:
```
inst myQueue_q of Queue
```
where the enclosing name myQueue is propagated to the Queue(...) function, so the internal q variable is correctly named.

But the implementation breaks down if you have nested hardware generators, since they would both see the same implicit enclosing name:
```
def MyGen1(mod: Module) = {
  val a = Module(new Thing1())
  // connect a with mod here or something and return one of their IO
}
def MyGen2() = {
  val b = Module(new Thing2())
  b
}

val myExample = MyGen1(MyGen2())
```
which would produce this:
```
  inst myExample_a of Thing1
  inst myExample_b of Thing2
```
even though `b` in `MyGen2` isn't directly assigned to `myExample`.

While this is a convoluted example, does something like this happen commonly, and is this acceptable (at least until a better solution is found)?

sdtwigg, 2016-10-04:
I'm in general not convinced that the benefits of the proposal under 'Functions' is worth the complexity, especially since the module annotation can be extended to introduce (unnested) names to the vals of the called functions anyway.

Since the proposal is blindly inserting implicit vals into code, you already outlined how this could fail. Except, unlike in 'normal' implicit-related debugging efforts, the user would just be pointed to a block of their own code (which has no implicits or functions calls using the referred implicits in it) and told there is an error. Plus, a nested val like that is a totally reasonable thing to do (which would work fine under the original proposal). Similarly, as you noted, the outlined version has a problem with the user's chained applys not working (except again, they would get references to trying to override implicits that are not at all in their stated function signature).

Something that was instead an annotation on the function definitions and wrapped all vals contained therein with a variant of TransitName rooted on the ultimate returned type seems more predictably reliable (and explainable to the end user). e.g.
```scala
@propagateCallerName
def createQueue(enq: DecoupledIO[T], depth: Int): DecoupledIO[T] = {
  val q = Module(???)
  q.io.enq <> enq
  q.io.deq
}
```
becomes
```scala
def createQueue(enq: DecoupledIO[T], depth: Int): DecoupledIO[T] = {
  val _toBeNamedBuffer = chisel.internal.NameBuffer // wraps a mutable.ListBuffer
  val q = _toBeNamedBuffer.add(Module(new Queue(...)), "q")
  q.io.enq <> enq
  _toBeNamedBuffer.resolve(q.io.deq)
}
```
A few notes:
* The add function just adds the argument to the buffer if subtype of HasId. Returns arg.
* The resolve simply TransitNames from the argument to all the items in the buffer. Returns arg.
* The TransitName functionality would need adjustment to ensure it can't go into an infinite loop (it currently doesn't bother to check) so you don't get messed up by 'val a = q.io.deq' in there or some other loop.
* The actual macro annotation would take the necessary 'fresh name' precautions (which is slightly  tricky with macro annotations but not impossible) for naming the _toBeNamedBuffer.
* The user also still has the option of just using @module (probably better called @addNames or something) to just wrap everything in Namer.
* Related to above, if we allow name suggestion priorities. The call to add could also register the suffix as a valid name to be used as-is in case the TransitName part is never triggered. I do have concerns about getting resolution order correct here when calls are nested.

This also should compose nicely with itself if, for example, a function with this annotation calls another the same annotation and so on.
In your example with "val b = MyGen1(MyGen2())", the MyGen2 fields wouldn't get any names (unless MyGen1 assigned its input parameter to some val). We could also toy with adding the input arguments to the buffer as well (perhaps at a 'lower priority' and with an extra prefix) in which case MyGen2's fields would get a name based off the name granted to MyGen1.

PS: Since the suggestion of wrapping all user-vals in an identity function came up: Note that, due to a reasonable scala typer feature decision, types and type parameters don't generally bind on the absolute narrowest type, which would be the singleton type of the passed argument (e.g. a.type or this.type). In some cases, where the desire for narrowing after the call is obvious, it will in fact work properly (tested in 2.11 with an identity function); however, it is hard to explain exactly when this is guarantee to work. Just something to keep in mind if using path-dependent or singleton types for anything.

ducky64, 2016-10-04:
So I'd actually like to avoid a TransitName-like solution if possible, because it separates the naming functionality into different places in both runtime and implementation location.

It seems to be possible to get some information about the "parent" of a macro application using `c.internal.enclosingOwner`, and some implementations (like [dsinfo](https://bitbucket.org/inkytonik/dsinfo/src/8df5dc58aae0ce6245c90b48fbdb877e61fca7d1/dsinfo/src/org/bitbucket/inkytonik/dsinfo/DSInfo.scala?at=default&fileviewer=file-view-default), which seems to do what we want, or close to it) seem to traverse the enclosing function for a matching val definition - though the performance implications are unclear. Still working out the details...

sdtwigg, 2016-10-04:
I am nervous about the use of enclosing* since the scalac team has already marked most of the functions in Enclosures as deprecated and scala.meta hasn't indicated any intent to resurrect the feature. It may be fine as long as you are only reading it since you have generally been able to extract the source files anyway.

ducky64, 2016-10-04:
I think the deprecation was more of a decision to clearly separate the stable / simple functionality from the less well thought out things that were still necessary to get things done (now part of c.internal.*). enclosingOwner isn't deprecated (yet?), though I'm still trying to figure out how to make good use of it.

sdtwigg, 2016-10-04:
My initial thoughts are that it could serve as an alternative to the annotation. (The Reg, Wire, etc. functions instead seek out names rather than return an unnamed thing.)
However, this doesn't work in the face of indirection:
```scala
def makeRegWithInit[T<:Data](init: T): T = Reg(init=init)
val myReg = makeRegWithInit(0.asUInt(32))
```
The annotation method would assign a name of myReg to the register whereas uses of enclosingOwner would at best discover makeRegWithInit.

I don't think uses of enclosingOwner are particularly helpful for the proposal under 'Functions' without rewriting the user's functions for them.

ducky64, 2016-10-04:
I think for your example the passthrough naming function would work, both because the internal Reg isn't declared as a val (and hence can't be named from makeRegWithInit's standpoint), and because the object would be returned to be named at a higher level (`val myReg = ...`)

An idea would be to use annotations to transform the function definition, like from
```
@functionNameTransform
def myFunction(val someParameter: Int) {
  val myObject = ...
}
```
to
```
def myFunction(args...) = macro AddNameParameter
def myFunctionImpl(enclosingName, args...) {
  val myObject = Named(enclosingName + "myObject", ...)
}
```
Where the AddNameParameter macro would rewrite the function invocation to add an additional name parameter as the first argument, if it's declared inside a val. Making this work well when the function invocation itself is transformed by a naming annotation would be a bit tricky (dependent on resolution order) but should be possible.
enclosingOwner also seems to suffer from the same problem as the initial Functions proposal, where nested function calls inside the same val definition returns that val term. Trying to find a non-deprecated way to get a parent Tree, on which it would be possible to do more advanced structural checks.

sdtwigg, 2016-10-04:
That example was assuming the proposal under 'Functions' was not available and you were just comparing the annotations approach to the enclosingOwner approach.

When I said "without rewriting the user's functions for them," I was implying that I do not think it is a good idea to use a macro to replace all of the user's function definitions with macro definitions. There are so many ways this can go wrong. (For example, named arguments don't work with macros and macros can't be used to fulfill abstract def contracts. Also, macros have very particular requirements on signature matching so it seems like you would need to further generate macro implementations, which then runs into separate compilation issues.)

According to the docs, it seems like you can continue calling enclosingOwner to get higher and higher trees although, again, I am extraordinarily concerned about the complexity and long-term implications of this.

ducky64, 2016-10-04:
So enclosingOwner returns a Symbol, and I don't (yet?) see (from the API docs) any way to get a Tree from a Symbol.

Macros do have their limitations, and these could be limitations of function naming for now. There's been attempts to fix named and default parameters, though none have been accepted into Scala yet. I don't think separate compile is an issue, since only the macro implementation needs to be compiled separately, and that can be generic inside chisel coreMacros.

That being said, while a TransitName like solution isn't the most elegant (with post name hooks and all that mess), it probably would be the most complete solution right now, at least given some of the issues with macros. However, it would be unable to handle cases where a function does not return a nameable object, though that's probably an uncommon case for hardware generators.

ducky64, 2016-10-17:
Ok, this was somewhat discussed at the last Chisel meeting, and the consensus was to focus on name propagation across functions first, then other nifty features (like variable interpolation).

New proposal, which is basically Stephen's thoughts, but:
- Instead of using TransitName, instead keep an IdentityHashMap of AnyRef to name (String) in the NameBuffer. This  avoids feature creep (like postname hooks) into core classes, limiting implementation to the base system.
- Related: Nix TransmitName completely, this supersedes it.
- Use of an AnyRef key (instead of depending on postname hook support in the class definition) allows a wider variety of objects to be returned without breaking name functionality. Not sure how often this is used, possibly with tuple return types?
- This avoids rewriting user functions as macros, which would have made this functionality incompatible with features like named or default variables.

However, this means it's now necessary to create and maintain links to the name buffer object from global context. This is probably best done within the structure of DynamicContext / Builder to centralize global state in case it becomes an issue (parallel compile?) in the future.

Thoughts?

sdtwigg, 2016-10-18:
What are you actually doing with the name buffer?
It isn't immediately clear how you translate the literally any elements in the name buffer with names into names for things you didn't touch via compile|runtime reflection. (You presumably have to store some sort of transiting information somewhere....)

Also, yes, rather than postname hooks involving arbitrary functions, you should instead just register a direct name linkage. That way you can walk the directed (potentially cyclic) graph of name linkages intelligently. (If a node has multiple candidate names due to multiple linkages, use the one on the shortest path from a root call.) Keeping the direct linkages and traversing with a BFS (with an 'already visited Set') should also allow you to avoid entering into loops in the cyclic graph case.

ducky64, 2016-10-18:
Idea is to have a stack of name buffers in a global data structure, and at the end of each function a name buffer is associated with the returned object.  In the calling function, when the returned object is named, the naming call also traverses the associated name buffer and names its contained elements by prefixing the name of the object to whatever name suffix was already in the buffer. Inefficient for deeply nested function calls, but simple.

An alternative (more efficient but also more complex) approach would be not recursively name at every stage but wait for the top-level and do one final name_everything call at the end, doing a graph-like traversal of the buffers. I don't think the efficiency gains are meaningful enough right now though.

I'm not sure what you're asking about how to translate elements into names for things not touched by reflection. If the naming macro (simple val transforms) can't name it, then it doesn't get named. More complex things like names for the result of arithmetic operations can come later.

ducky64, 2016-10-18:
I found a nice clean solution to avoid unnecessary recursive naming calls (opting for the graph-traversal-like method instead) and it's implemented.

ducky64, 2016-12-07:
@sdtwigg to reiterate the discussion from today, the desired changes are (and please correct me if my interpretations are wrong):
- Have an additional annotation for blind (non-hierarchical) naming in library functions.
- chiselName still defaults to hierarchical naming. Perhaps have checking such that annotating a non-returning function with chiselName will error out.
- For all naming annotations, in the case of duplicate names, the first one takes priority (alternatively, this could also be "the first one in the highest scope", though that may be more confusing). This would be a compromise between full naming that downstream tools can rely on and simple naming, in that users could rename things if they wanted unique name references.
- Module naming annotation should recurse into functions. The default will be to apply blind naming (alternatively, this could apply hierarchical naming where a AnyRef return type is defined, falling back to blind naming otherwise)
- Documentation for users to override annotation names. Probably by exposing the Named(...) construct?
- Don't touch TransitName for now. The two systems will live side-by-side for now.

sdtwigg, 2016-12-14:
The main thing I don't fully understand is why you would expose Named rather than using suggestName (for blind naming) or TransitName (for propagated naming).

ducky64, 2016-12-14:
The exact mechanism is up for discussion (I didn't implement either in the PR; it would be a follow-on change especially since it doesn't affect the naming annotation API), we can try to reach a consensus at the meeting today.
I'm also not sure I like TransitName because it heavily relies on having special support inside the Chisel core classes. In the end, what gets exposed to the user is ultimately a frontend / API question, possibly independent of the underlying implementation.

closed 2017-01-26 by ducky64