chipsalliance/chisel #305: Not bound to synthesizable node?

donggyukim, 2016-09-30:
I got an error with a previously working design as follows:

[info]   chisel3.core.Binding$BindingException: 'that' (chisel3.util.DecoupledIO@42f).bits: Not bound to synthesizable node, currently only Type description
[info]   at chisel3.core.Binding$.checkSynthesizable(Binding.scala:155)
[info]   at chisel3.core.Data.bulkConnect(Data.scala:142)
[info]   at chisel3.core.Data.$less$greater(Data.scala:170)
[info]   at strober.Channel.<init>(sim.scala:62)
[info]   at strober.SimUtils$$anonfun$genChannels$1$$anonfun$1.apply(util.scala:47)
[info]   at strober.SimUtils$$anonfun$genChannels$1$$anonfun$1.apply(util.scala:47)
[info]   at chisel3.core.Module$.do_apply(Module.scala:29)
[info]   at strober.SimUtils$$anonfun$genChannels$1.apply(util.scala:47)
[info]   at strober.SimUtils$$anonfun$genChannels$1.apply(util.scala:45)
[info]   at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)

What am I supposed to do to fix this?

aswaterman, 2016-09-30:
If it previously emitted working FIRRTL, this is almost certainly a chisel3 bug.

ucbjrl, 2016-09-30:
Does adding an:

    import chisel3.core.ExplicitCompileOptions.NotStrict

to the file sim.scala suppress the error?

aswaterman, 2016-09-30:
er, right, I was assuming strober only used import Chisel._, not chisel3._.

donggyukim, 2016-09-30:
It turns out that io definitions without `IO` can cause these wired errors. I think there should be warnings for not using it.

aswaterman, 2016-09-30:
@donggyukim was that with chisel3 or Chisel?  In Chisel, IO is not supposed to be required, so it is a bug in Chisel if it issues the error.  If it was chisel3, you need IO.

donggyukim, 2016-09-30:
It is with chisel3. Then, there should be errors without `IO`s.

aswaterman, 2016-09-30:
ok cool

aswaterman, 2016-09-30:
I assume issuing a warning is fundamentally difficult.  If it were easy to detect something was an IO without IO(), then I guess we wouldn't need IO()

ucbjrl, 2016-09-30:
Right. We could improve the error message. In order to handle the Chisel2, IO-not-required case, we check to see if the unbound element is a component of the module's io. If so, we could issue a more useful error message.

commit 1abe52da (2016-10-06): Generate a better error message for missing IO() wrapper - fix #305
commit 7aea39d4 (2016-10-06): Merge pull request #312 from ucb-bar/improveunboundioerrormessage305

Generate a better error message for missing IO() wrapper - fix #305
closed 2016-10-06 by ucbjrl