chipsalliance/chisel #309: Documenting proper ChiselMain replacement

shunshou, 2016-10-03:
So in Chisel2, the Verilog file was generated via ChiselMain. I believe rocket-chip still uses ChiselMain.run to generate Chirrtl, and then the Firrtl executable is separately run via makefile to generate Verilog. But if rocket-chip isn't our top module -- as in some DSP stuff I'm starting to work on, what exactly is supposed to be the standardized way to do this in Chisel3, now that ChiselMain.run is supposed to be deprecated?

---

As far as I can tell, the new Chisel3 TesterDriver does go through the process of generating verilog, as evidenced by the content of test_run_dir. If that's the case, are we just supposed to assume we have to run sbt test to generate Verilog?

---

Alternatively, we can do what ChiselMain.run does to generate a Chirrtl file and then use Driver.firrtlToVerilog to generate the Verilog file via the Scala ProcessBuilder. I know people have been talking about running Firrtl directly in Chisel, but the Driver.firrtlToVerilog function still requires an input file as opposed to being able to directly take in some ir.circuit. 

If the latter option is supposed to become standardized, it'd be great to have that documented somewhere. Note that chisel-template doesn't have a main function anywhere, so only sbt test works and not sbt run. 

I basically added something like:

```
object test {
  import java.io._
  import scala.sys.process._

  def firrtlToVerilog(prefix: String, dir: File): ProcessBuilder = {
    Process(
      Seq("./firrtl",
          "-i", s"$prefix.fir",
          "-o", s"$prefix.v",
          "-X", "verilog"),
      dir)
  }

  def main(args: Array[String]): Unit = {
    val gen = () => new GCD
    val circuit = Driver.elaborate(gen)
    Driver.parseArgs(args)
    val output_fir_file = new File(Driver.targetDir + "/" + circuit.name + ".fir")
    Driver.dumpFirrtl(circuit, Option(output_fir_file))
    val firrtl_dir = new File(Driver.targetDir + "/" + "dsp-framework/firrtl/utils/bin/")
    val genV = firrtlToVerilog(Driver.targetDir + "/" +circuit.name,firrtl_dir)
    genV.run()
  }
}
```

To the end of examples/GCD.scala in chisel-template to generate verilog with sbt run. 

Two things to note: 

I had to dig around to find out that firrtlToVerilog existed + generated a ProcessBuilder. I had to dig around more to find out that firrtl would only be run if I did genV.run()... I think that should really be documented.

The other thing is, I'm not sure why I needed ./firrtl instead of just firrtl in the Process... so I just made my own firrtlToVerilog in the test obj... 

---

Another alternative, and likely the one people actually plan to use:

After a *.fir file has been dumped:

```
    val lines = scala.io.Source.fromFile(Driver.targetDir + "/" + circuit.name + ".fir").mkString
    val annotationMap = firrtl.Annotations.AnnotationMap(Nil)
    val writer = new java.io.StringWriter
    val compiler = new firrtl.VerilogCompiler()
    compiler.compile(firrtl.Parser.parse(lines), annotationMap, writer)
```

And then print writer contents to file.

The main issue with this is still that I need to print the Chirrtl to an intermediate file before using in Firrtl. Otherwise, I get:

found   : chisel3.internal.firrtl.Circuit
[error]  required: firrtl.ir.Circuit

Is there some kind of helper function in Firrtl that I'm not aware of that casts chisel3.internal.firrtl.Circuit to firrt.ir.Circuit? 

---

Anyways, I'm more or less ok with all of the above or none of the above, but would like to know what people actually want to standardize...

chick, 2017-01-12:
@shunshou I have added an entry to the chisel3 FAQ at [I just want some verilog, What do I do?](#get-me-verilog)
There is also a stackoverflow answer for this at [How to generator verilog from a chisel module](http://stackoverflow.com/questions/40470153/is-there-a-simple-example-of-how-to-generate-verilog-from-chisel3-module)
If this is not what you meant or it needs enhancement or better SEO please let me know

shunshou, 2017-01-12:
How about more specific questions:

How do I generate Verilog in a directory of choice X?
How do I pass in annotations for a FIRRTL pass aka replseqmem? Or other
more general but useful annotations

chick, 2017-02-02:
@shunshou I added the answer to the first question to the how do I just get verilog faq entry
In general annotations get added because they are generated in the modules using an API determined by the library writer.  As the latest DspTools does in order to move verilog to output directory when using verilator.  There is more writeup on Annotations in the short users quide to using chisel under [Annotations](https://github.com/ucb-bar/chisel3/wiki/Annotations%20Extending%20Chisel%20and%20Firrtl).  The replseqmem I don't understand, but I believe that we now have all the machinery to make replseqmem passed as an annotation and I believe this is the way @azidar and @jackkoenig want it to be.  If it isn't right yet let's put it on the hacking to do list

shunshou, 2017-02-02:
I think it works. Thanks!

closed 2017-02-02 by chick