chipsalliance/chisel #324: Reg(init = Vec( ... )) generates invalid FIRRTL

jwright6323, 2016-10-14:
Similar to #230 I am getting invalid FIRRTL when trying to use Reg(init = Vec( ... )). However, unlike #230 the init value is a literal. The error can be resolved by wrapping the Vec in a Wire().

example:

`val errorCounts = Reg(init = Vec(bertNumWays, UInt(0, width = bertCountWidth))) `

I noticed this with both Bools and UInts, if that helps. Here is a stack trace:

`Exception in thread "main" firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[:TestHarness.DefaultHUpTopConfig.fir@83211.4]: [module Bert] Reference T_21 is not declared.
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlE$1(CheckChirrtl.scala:101)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$3.apply(CheckChirrtl.scala:130)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$3.apply(CheckChirrtl.scala:130)
	at firrtl.ir.DefRegister.mapExpr(IR.scala:163)
	at firrtl.Mappers$StmtMagnet$$anon$3.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:55)
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlS$1(CheckChirrtl.scala:130)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$5.apply(CheckChirrtl.scala:131)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$5.apply(CheckChirrtl.scala:131)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.ir.Block.mapStmt(IR.scala:227)
	at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:41)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:55)
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlS$1(CheckChirrtl.scala:131)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlM$1$3.apply(CheckChirrtl.scala:144)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlM$1$3.apply(CheckChirrtl.scala:144)
	at firrtl.ir.Module.mapStmt(IR.scala:433)
	at firrtl.Mappers$ModuleMagnet$$anon$11.map(Mappers.scala:115)
	at firrtl.Mappers$ModuleMap$.map$extension(Mappers.scala:125)
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlM$1(CheckChirrtl.scala:144)
	at firrtl.passes.CheckChirrtl$$anonfun$run$1.apply(CheckChirrtl.scala:147)
	at firrtl.passes.CheckChirrtl$$anonfun$run$1.apply(CheckChirrtl.scala:147)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at firrtl.passes.CheckChirrtl$.run(CheckChirrtl.scala:147)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:47)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:47)
	at firrtl.Utils$.time(Utils.scala:55)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:47)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:45)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.SimpleRun$class.run(LoweringCompilers.scala:45)
	at firrtl.Chisel3ToHighFirrtl.run(LoweringCompilers.scala:61)
	at firrtl.Chisel3ToHighFirrtl.execute(LoweringCompilers.scala:68)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:71)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:70)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.Compiler$class.compile(Compiler.scala:70)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:198)
	at firrtl.Driver$.compile(Driver.scala:109)
	at firrtl.Driver$.run(Driver.scala:183)
	at firrtl.Driver$.main(Driver.scala:86)
	at firrtl.Driver.main(Driver.scala)`

aswaterman, 2016-10-14:
The problem is that Vec(n, t) yields a type, not a value.  Vec.fill(bertNumWays)(UInt(0, width = bertCountWidth)) is what you want.

jwright6323, 2016-10-14:
That makes sense, thanks.

closed 2016-10-14 by jwright6323