chipsalliance/chisel #326: Vec of Modules Allows User to "assign to" Output

seldridge, 2016-10-17:
The output of a vector of modules can be inadvertently overwritten by last-connect semantics if a user mixes up input and output declarations. Plain instantiations of Modules are able to check and provide protection against this.

Note: this is clearly user error. However, it gets confusing once you have nested Bundles with Decoupled/Valid (which was the original, non-MWE problem).

The following example illustrates this. I have an output `x` (which should be an INPUT) and I overwrite it's assignment without an error.
```scala
import Chisel._

class Sub extends Module {
  val io = new Bundle {
    val x = Bool(OUTPUT)
  }
}

class VecSub(n: Int = 1) extends Module {
  val io = new Bundle {
    val y = Bool(INPUT)
  }
  val s = Vec.fill(n){ Module(new Sub).io }

  for (i <- 0 until n) {
    s(i).x := io.y
  }
}

class VecSub1 extends VecSub
```

This results in the following FIRRTL (last two lines are what matters here):
```text
;buildInfoPackage: chisel3, version: 3.1-SNAPSHOT, scalaVersion: 2.11.7, sbtVersion: 0.13.7, builtAtString: 2016-10-17 22:40:42.235, builtAtMillis: 1476744042235
circuit VecSub1 : 
  module Sub : 
    input clock : Clock
    input reset : UInt<1>
    output io : {x : UInt<1>}
    
    io is invalid
    
  module VecSub1 : 
    input clock : Clock
    input reset : UInt<1>
    output io : {flip y : UInt<1>}
    
    io is invalid
    inst Sub_1 of Sub @[PeLike.scala 15:30]
    Sub_1.io is invalid
    Sub_1.clock <= clock
    Sub_1.reset <= reset
    wire s : {x : UInt<1>}[1] @[PeLike.scala 15:22]
    s is invalid @[PeLike.scala 15:22]
    s[0].x <= Sub_1.io.x @[PeLike.scala 15:22]
    s[0].x <= io.y @[PeLike.scala 18:12]
```

This seems particularly nasty as I can't think how one would catch this without detecting some allowable last-connect construct.

Chisel3 hash: ed872df330cb7dfabdd0e0866176f8f5be8861da

ucbjrl, 2016-10-17:
@seldridge , could you try this with an `import chisel3._` instead of `import Chisel._`? This enables stricter connection checking.

seldridge, 2016-10-17:
Thanks for the response! Unfortunately, switching to `chisel3._` and adding the `IO` wrappers gives me the same thing. :(

This is what I'm trying: 
```scala
import chisel3._

class Sub extends Module {
  val io = IO(new Bundle {
    val x = Bool(OUTPUT)
  })
}

class VecSub(n: Int = 1) extends Module {
  val io = IO(new Bundle {
    val y = Bool(INPUT)
  })
  val s = Vec.fill(n){ Module(new Sub).io }

  for (i <- 0 until n) {
    s(i).x := io.y
  }
}

class VecSub1 extends VecSub
```

Honestly, I'd be very surprised if anything caught this, as I'm effectively assigning the IO of a module to a wire. Once I'm done that assignment, I should be able to last-connect away with that wire. However, this just isn't what I intended... as I really want a Vec of IO and not a Vec of Wires hooked up to the IO of a module.

Tangentially, @ucbjrl, what's the guidance on using `Chisel._` vs. `chisel3._`? I noticed that the latter was much stricter, but introduced all sorts of requirements that I'd never seen before (like `IO` wrappers) and I backed off of this as I didn't see rocket-chip using it. Is this something that is advisable to be using?

ucbjrl, 2016-10-17:
We encourage new designs to use `import chisel3._`. The `Chisel` package attempts to provide a compatibility layer with existing chisel2 designs. A substantial missing piece is the documentation, which should be available in the very near future.

sdtwigg, 2016-10-18:
The underlying problem here is a misdocumentation of what  "Vec.fill(n){ Module(new Sub).io }" is actually doing.

It doesn't return a Vec of Modules, or even a Vec of Module IOs. It returns a Wire that is a type of Vec(n, [the widened type of Sub.io.type]) that has already been connected to the actual IO each corresponding Module Sub. Most versions of chisel3 will do a bidirectional connection, so module inputs are actually driven by the wire and outputs are driving those parts of the Wire. Since it is merely a Wire, you are free to reconnect any signal, including the ones already being driven by the output.

seldridge, 2016-10-18:
Thanks, @sdtwigg. Yes, that's the conclusion that I came to. What I'm more getting at is how to avoid this or if the whole approach of `Vec.fill(n){ Module(...).io }` should be rethought? While the above example is clearly user error, things get more complicated when this issue is stacked with, e.g, #148. In the non-MWE originating problem, I was declaring signals of a nested Bundle as `INPUT` and the `.asOutput` of `QueueIO` was failing. The end result is that these signals were given the wrong direction and the `Wire`, which I rightly get back from `Vec.fill`, effectively strips all sane checking on how this signal is being used.

It's possible that using the stricter checking of `chisel3._` globally that @ucbjrl suggests would have caught this before it gets to this point, but I haven't had a chance to try that, yet (I'm in Chisel2 -> Chisel3 porting hell right now and doing this incrementally).

Relatedly, using a `Seq[Module]` is an option. This provides checking of compile-time connections, but won't help if doing some Mux structure (selecting a module's IO using a `UInt`). This requires making a `Wire(Vec(...))` of the IO and I'm back at the same place.

ben-k, 2016-10-19:
@seldridge, amen to your last though.  In Chisel3, using `Seq` instead of `Vec` unless you absolutely have to dynamically index will save you a lot of pain.

ducky64, 2017-02-06:
We think that this isn't a problem anymore in `chisel3._`.
If it still is a problem, re-open this issue.

closed 2017-02-06 by ducky64