chipsalliance/chisel #331: Use verilator VPI interface

ucbjrl, 2016-10-19:
Predicting verilator's signal names is increasingly problematic. #329 (compiling with -O0) breaks the existing harness communication due to signal renaming.

jackkoenig, 2016-10-20:
Which harness breaks? The chisel3 tests pass using their testing harness, is it a problem in https://github.com/ucb-bar/chisel-testers?

donggyukim, 2016-10-20:
We should move this to chisel-tester.

wallento, 2016-11-04:
Hi, actually I think you want to use DPI for it instead of VPI.

donggyukim, 2016-11-04:
@wallento Can we access internal signals with DPI?

wallento, 2016-11-04:
Yes, you can. It depends a bit on what you want to achieve. Do I understand it correctly it is driving signals in a testbench? Then DPI works fine, VPI has the only advantage that you can mess in the entire hierarchy. DPI instead is the suggested way for ease of use and clarity from my point of view.

wallento, 2016-11-04:
If you want to randomly access all signals than VPI (or plain C++) and `/* verilator public */` are indeed needed.

oharboe, 2019-12-04:
I created a Perl post-processing hack to inject ```/* verilator public */``` after input/output signals that I need to reach from C++. This hack broke with 3.2.0, because the Verilog is very slightly different in 3.2.0.

It would be great if there was a robust way to specify that should ```/* verilator public */``` follow input or output signals to a module in the generated Verilog.

seldridge, 2019-12-04:
@oharboe: I don't have an example of doing this, but the [`DescriptionAnnotation`](https://www.chisel-lang.org/api/firrtl/latest/firrtl/DescriptionAnnotation.html) is likely the stable way of doing this coupled with https://github.com/freechipsproject/firrtl/pull/1172. The attributes you're talking about are generally needed...

wallento, 2019-12-04:
just a quick update on the verilator side: There will soon be support to specify public signals to be accessed via VPI in a separate file. Cheers!

jackkoenig, 2019-12-04:
Thanks for your hard work on this @wallento 🙂