chipsalliance/chisel #336: Seq as alternative to Vec , Vec bugs

shunshou, 2016-10-21:
For things that 

1) Don't need to be dynamically accessed 
2) Support multiple bitwidths (for the purposes of compile-time generability, you want to be able to have some arbitrarily long list of signalName_index)

@jackkoenig said that Seq's of Chisel data types were supposed to give better names now, and that might be the solution to the Bundle of arbitrary length I wanted earlier for non-homogeneous Vecs.

A couple things I tried:

First, I wanted the equivalent of vec.asUInt (or .toBits). 

It seems like the functionality exists [here](https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/SeqUtils.scalal) BUT it's private to Chisel... 

How would you support the equivalent of cloneType?

---

Bugs:

In an IO bundle, I have:

`val test = Vec(Seq.fill(3)(Bits(OUTPUT, Width(3)))) `

In the body of my module, I have:

`io.test(1) := Vec(Seq.fill(3)(state === UInt.Lit(0))).asUInt`

This results in the following Chisel compile error

```
[error] (run-main-0) scala.MatchError: null
scala.MatchError: null
	at chisel3.core.Binding$.chisel3$core$Binding$$walkToBinding(Binding.scala:62)
	at chisel3.core.Binding$.bind(Binding.scala:88)
	at chisel3.core.Module.Port(Module.scala:64)
	at chisel3.core.Module.IO(Module.scala:89)
	at chisel3.core.Module.computePorts(Module.scala:138)
	at chisel3.core.Binding$.chisel3$core$Binding$$elementOfIO$1(Binding.scala:128)
	at chisel3.core.Binding$$anonfun$checkSynthesizable$1.apply(Binding.scala:172)
	at chisel3.core.Binding$$anonfun$checkSynthesizable$1.apply(Binding.scala:167)
	at chisel3.core.Binding$.chisel3$core$Binding$$walkToBinding(Binding.scala:63)
	at chisel3.core.Binding$.checkSynthesizable(Binding.scala:165)
	at chisel3.core.Data.bulkConnect(Data.scala:156)
	at chisel3.core.Vec$.chisel3$core$Vec$$doConnect$1(Aggregate.scala:68)
	at chisel3.core.Vec$$anonfun$do_apply$3.apply(Aggregate.scala:74)
	at chisel3.core.Vec$$anonfun$do_apply$3.apply(Aggregate.scala:73)
	at scala.collection.TraversableLike$WithFilter$$anonfun$foreach$1.apply(TraversableLike.scala:778)
	at scala.collection.Iterator$class.foreach(Iterator.scala:742)
	at scala.collection.AbstractIterator.foreach(Iterator.scala:1194)
	at scala.collection.IterableLike$class.foreach(IterableLike.scala:72)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
	at scala.collection.TraversableLike$WithFilter.foreach(TraversableLike.scala:777)
	at chisel3.core.Vec$.do_apply(Aggregate.scala:73)
```

If I replace the port in the IO bundle with:

`val test = Vec(3,Bits(OUTPUT, Width(3))) `

Stuff compiles...

---

The above is maybe a bad example b/c the elemets of the Vec are the same width, but assuming that they aren't, I'd definitely only want to allocate enough bits for each IO port... I tried to do the same with Seq inside the bundle with the hope that it'd do smart naming with reflection:

Inside the bundle:

`val test = Seq.fill(3)(Bits(OUTPUT, Width(3)))`

Inside the module body:

`io.test(1) := UInt.Lit(0)`

Firrtl crashes with:

```
[error] (run-main-0) firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[SynchronousSARCtrl.scala 135:27]: [module SynchronousSARCtrl] Reference T_9 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[SynchronousSARCtrl.scala 135:27]: [module SynchronousSARCtrl] Reference T_9 is not declared.
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlE$1(CheckChirrtl.scala:101)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$3.apply(CheckChirrtl.scala:130)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$3.apply(CheckChirrtl.scala:130)
	at firrtl.ir.Connect.mapExpr(IR.scala:242)
	at firrtl.Mappers$StmtMagnet$$anon$2.map(Mappers.scala:44)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:55)
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlS$1(CheckChirrtl.scala:130)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$5.apply(CheckChirrtl.scala:131)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlS$1$5.apply(CheckChirrtl.scala:131)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.ir.Block.mapStmt(IR.scala:227)
	at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:41)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:55)
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlS$1(CheckChirrtl.scala:131)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlM$1$3.apply(CheckChirrtl.scala:144)
	at firrtl.passes.CheckChirrtl$$anonfun$firrtl$passes$CheckChirrtl$$checkChirrtlM$1$3.apply(CheckChirrtl.scala:144)
	at firrtl.ir.Module.mapStmt(IR.scala:443)
	at firrtl.Mappers$ModuleMagnet$$anon$11.map(Mappers.scala:115)
	at firrtl.Mappers$ModuleMap$.map$extension(Mappers.scala:125)
	at firrtl.passes.CheckChirrtl$.firrtl$passes$CheckChirrtl$$checkChirrtlM$1(CheckChirrtl.scala:144)
	at firrtl.passes.CheckChirrtl$$anonfun$run$1.apply(CheckChirrtl.scala:147)
	at firrtl.passes.CheckChirrtl$$anonfun$run$1.apply(CheckChirrtl.scala:147)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at firrtl.passes.CheckChirrtl$.run(CheckChirrtl.scala:147)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:47)
	at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:47)
	at firrtl.Utils$.time(Utils.scala:56)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:47)
	at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:45)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.SimpleRun$class.run(LoweringCompilers.scala:45)
	at firrtl.Chisel3ToHighFirrtl.run(LoweringCompilers.scala:61)
	at firrtl.Chisel3ToHighFirrtl.execute(LoweringCompilers.scala:68)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:71)
	at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:70)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.Compiler$class.compile(Compiler.scala:70)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:201)
```

== You can't have a Seq in a Bundle and therefore need some other mechanism to support heterogenous non-dynamically accessed Vec-like things?

sdtwigg, 2016-10-21:
I fiddled with this a bit.
```scala
val test = Vec(Seq.fill(3)(Bits(OUTPUT, Width(3))))
```

is wrong, particularly when used in a Bundle def, but should have been registering a better error message.  I just checked the code sample on a less complex variant of the chisel3 that does not try to do the compatibility layering and it returns the following error message:
> Chisel.internal.Binding$BindingException: 'that' (Chisel.UInt@####): Not bound to synthesizable node, currently only Type description

The stack trace points to that line as the first user code point, but we can also see the problem is in the Vec.apply when it tries to do a connection. This actually points to a major UX problem with apply that I will now delve into....

First, we must ask ourself what the 'true type signature' of Vec.apply(in: Seq[T<:Data]) is. A key realization is that ALL Chisel.Data and subtypes are overloaded and conflate 2+ 'true-types' (partly due to the mutable state inside of them that toggles and then does something). A Data is either 'unbound' and thus is just representing a potential hardware type description that can be used to emit some register, wire, or port declaration. These are created with e.g. UInt(width=4) or Vec.apply(Int, T<:Data) and can only be used to create other unbound type descriptions or passed into a Reg, Wire, or IO. An interesting observation is that instances of these true-types are somewhat interchangeable. For example, it doesn't matter (except for performance) if you accidentally call chiselCloneType an extra time on a gen before using it anywhere. (This also excludes bad code relying on referential equality for unbound types.)

The other true types (depending on if you count each as a distinct true-type or not) is what you get from the result of Reg, Wire, and IO. Instances of these true-types, which I call 'synthesizable or bound' in error messages, can actually be used to describe hardware operations. You can (for different sets of them) assign them, add them, subtract them, do bit conversions between then, etc. Also, all those operations (that return something) return another bound true-type. Similarly, you generally shouldn't use these true-types in places where you could use an unbound type:
Reg(0.asUInt) is technically valid but equivalent to Reg(UInt()), not Reg(init=0.asUInt), as commonly expected. Note further that, since each instance of these true-types are bound to a specific hardware value, they are rarely 'interchangeable' without changing the underlying circuit.

Note, there is one particularly important case where you CAN'T use these values where an unbound-type is required: Bundle definitions. Bundle definitions are always assumed to be describing the unbound type. These definitions are then passed into and IO, Reg or Wire in order to materialize it in the graph. Chisel just doesn't know how to handle a 'partially-bound' type. For example, if you pass a Bundle that defines a field = Reg(...) into Wire, its implementation just has no way of coercing that field to now represent a Wire. (Also, since it WAS a Reg or Wire, what happens to already-emitted operations on that field? Should they point to the new rebound type? But what if that violates user expectations?). In a sense, Bundle definitions are like C-structs, they just represent the potential structure of data but you can't give them defaults.

The reason I elaborated that point so deeply is that understanding this deficiency in Chisel's use of the type system really helps drill down on what the error is. Also, this subtlety really outlines why Vec.apply can be so confusing, especially for new-comers. The different forms all seem to take and return the same type, but really do and return substantially different things.

Vec.apply(Int, T<:Data): Vec[T] both takes an unbound type of scala-type T and returns an unbound type of scala-type Vec[T] that can be used to instantiate Reg, Wire, etc. that is a hardware array of that value T. This function does not actually create hardware.

In contrast, Vec.apply(Seq[T<:Data]): Vec[T] both takes a BOUND type of scala-type Seq[T] and returns a bound Wire of scala-type Vec[T]. This function is creating hardware in the current Module: it creating a Wire(Vec(Seq.length, T)) and then connecting it to the input Data. Hence, why both the input and result must be bound to actual hardware.

Looking at your lines in the context of the prior discussion:
```scala
val test = Vec(Seq.fill(3)(Bits(OUTPUT, Width(3)))) // in a Bundle definition
```
The call to Vec.apply is the Seq overload and thus expects a bound type (since it is connecting it to something). However, Bits(OUTPUT, Width(3)) represents an unbound type (it isn't on the hardware graph) and thus Chisel complains at runtime since you can't emit operations referencing an unbound type.

Further, even if we fixed this so the call to Seq.fill was fixed to be given a bound type, the circuit should still fail to elaborate because the return of the Vec is bound. Thus, the IO call will complain that field test is already bound to hardware and thus can't be part of representing an IO.

Note that the null match error is due to a bug in the compatibility error triggered specifically because this Bundle is part of the io definition. The compatibility layer will try to simulate wrapping your io in IO for you (thus marking all the fields as being bound) by looking at the io of the currentModule. However, this issue is occuring due to the  [... cut ...]
This desire is why I advocate (at least as a 'super-user' level feature) the ability to define a Bundle that doesn't use reflection to determine its elements, instead allowing the user to specify the namedElts mapping of String->Data. In your case you would have like:
```scala
override val namedElts = ListMap(
  (for((elem, idx) <- test.zipWithIndex) yield s"test_$idx" -> elem):_*
)
```
Sure, Seq might be popular enough to warrant inclusion in the actual reflection (excepting the compatibility concerns) just like Option was, but who knows what the next one will be. The problem is that the user is just stuck with the default reflection behavior with no way of fixing it.

shunshou, 2016-10-21:
Yeah, sorry, I didn't bother looking for the root causes of these issues. Thanks for spending time to delve into this. 

I guess I don't necessarily care what the final solution is -- Vec is just currently really confusing to use [and I'm definitely not the only one with these problems], and I just really want some way to be able to programmatically add elements of varying widths (possibly types) based off of some generator parameter.

I agree that there should be a more general solution than special casing Seq and Option...

Also, there should be some better way to show noob chiselers what to do/not do as far as which apply methods to use [too many options = bad]. Having to sift through Chisel code is painful. .__.

sdtwigg, 2016-10-21:
On the specific matter of the apply methods (both for Vec and UInt), I am still insistent that they are a bad design in the API (regardless of their 'legacy status') since they exhibit such different behaviors even with similar names, argument, and return types.

The fact that you are not the first, nor will be the last, to have this issue reinforces my complaint.

ben-k, 2016-10-21:
There are a large number of outstanding issues around the Chisel3 Vec (e.g., #247, #260).  I strongly agree that some of the apply methods need to be deprecated, and we need to standardize around the "right" way to use Vec, with clear documentation of what it should and should not be used for.

ducky64, 2017-02-06:
Superseded by #445 which will have stricter type / hardware node checks.
Heterogeneous vecs are unsupported, recommended alternative is to use Record.

closed 2017-02-06 by ducky64