chipsalliance/chisel #348: Weird Regression with 224ee39 (Vec issue)

seldridge, 2016-11-02:
I'm getting a weird regression with 224ee39 (checked via bisect), but I haven't been able to get an MWE that actually trips. I have a `Vec(Seq.fill(n)(Module(new MyMod).io))` and repeated assignment with more than 4 statements causes the reflection to go nuts.

The exact structure is the following (from ripping out all the extraneous code of an internal version of the [original](https://github.com/bu-icsg/xfiles-dana/blob/master/src/main/scala/ProcessingElementTable.scala#L137)):
```scala
class ProcessingElementTableBase(implicit p: Parameters) extends DanaModule()(p) {
  val io = IO(new PETableInterface)

  val pe = Vec(Seq.fill(p(PeTableNumEntries))(Module(new ProcessingElementLearn).io))

  pe(UInt(0)).req.valid := Bool(true)
  pe(UInt(0)).req.valid := Bool(true)
  pe(UInt(0)).req.valid := Bool(true)
  pe(UInt(0)).req.valid := Bool(true)
  // The next line (148) breaks it
  pe(UInt(0)).req.valid := Bool(true)
}

// Line 159 is the "extends" part here
class ProcessingElementTableLearn(implicit p: Parameters)
    extends ProcessingElementTableBase()(p)
```

```
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:29)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:190)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:190)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:194)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:192)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:192)
	at chisel3.Driver$.elaborate(Driver.scala:190)
Caused by: java.lang.reflect.InvocationTargetException
	at sun.reflect.GeneratedConstructorAccessor49.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at util.ParameterizedBundle.cloneType(Misc.scala:10)
	at util.ParameterizedBundle.cloneType(Misc.scala:7)
	at chisel3.core.Aggregate.cloneTypeWidth(Aggregate.scala:18)
	at chisel3.core.Aggregate.cloneTypeWidth(Aggregate.scala:17)
	at chisel3.core.Vec$$anonfun$3.apply(Aggregate.scala:69)
	at chisel3.core.Vec$$anonfun$3.apply(Aggregate.scala:69)
	at chisel3.core.Vec.apply(Aggregate.scala:195)
	at dana.ProcessingElementTableBase.<init>(ProcessingElementTable.scala:148)
	at dana.ProcessingElementTableLearn.<init>(ProcessingElementTable.scala:155)
	at dana.Dana$$anonfun$7.apply(Dana.scala:264)
	at dana.Dana$$anonfun$7.apply(Dana.scala:264)
Caused by: java.lang.ClassNotFoundException: sun.reflect.GeneratedConstructorAccessor49
	at java.net.URLClassLoader$1.run(URLClassLoader.java:366)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:195)
	at chisel3.internal.ErrorLog.chisel3$internal$ErrorLog$$isUserCode$1(Error.scala:53)
	at chisel3.internal.ErrorLog$$anonfun$1.apply(Error.scala:56)
	at chisel3.internal.ErrorLog$$anonfun$1.apply(Error.scala:56)
	at scala.collection.IndexedSeqOptimized$$anonfun$indexWhere$1.apply(IndexedSeqOptimized.scala:204)
	at scala.collection.IndexedSeqOptimized$$anonfun$indexWhere$1.apply(IndexedSeqOptimized.scala:204)
	at scala.collection.IndexedSeqOptimized$class.segmentLength(IndexedSeqOptimized.scala:195)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2077)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
```

However, since I can't get an equivalent MWE failing in a less cluttered environment, the issue could be related to some interaction with something else...

seldridge, 2016-11-02:
A few more details (which are technically just interpretations of/elaborations on the stack trace). I'm importing `chisel3._` in the above module. The `req` interface is decoupled and its data inherits from `ParameterizedBundle` of `rocketcihp.util` which imports `Chisel._` (possibly related to @aswaterman's warning in 224ee397db324067b35874bf22c76a63e9ca531b).

After digging a bit, it looks like  `this.getClass.getConstructors.head.newInstance(p).asInstanceOf[this.type]` in [ParameterizedBundle](https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/util/Misc.scala#L10) is barfing with a `java.lang.ClassNotFoundException` (checked by catching the exception in `ParameterizedBundle` and running `getTargetException` on it):
```
Caused by: java.lang.Exception: WTF? java.lang.ClassNotFoundException: sun.reflect.GeneratedConstructorAccessor15
	at chisel3.compatibility.throwException$.apply(throwException.scala:11)
	at util.ParameterizedBundle.cloneType(Misc.scala:18)
	at util.ParameterizedBundle.cloneType(Misc.scala:7)
```

I'm at a loss as to why this is dependent on the number of times that I run the dynamically indexed read method of `Vec`, however...

seldridge, 2016-11-02:
And, I have a rough fix: The file declaring the class for `req` needs to use the compatibility layer and `import Chisel._`.

If somebody gets the time, could you fill me in on what's likely going on here?

I do realize I've been living precariously on the edge of advised/sane adoption of alpha (now beta!) tools, but I appreciate all the help.

colinschmidt, 2017-02-07:
I was unable to reproduce this with simple types in current chisel3

```
  val adders = Vec(Seq.fill(4)(Module(new FullAdder).io))                                                                                                     
  adders(UInt(0)).cin := UInt(1)                                                                                                                              
  adders(UInt(0)).cin := UInt(1)                                                                                                                              
  adders(UInt(0)).cin := UInt(1)                                                                                                                              
  adders(UInt(0)).cin := UInt(1)
  adders(UInt(0)).cin := UInt(1) 
```
works fine.

jackkoenig, 2017-02-07:
@seldridge I'm going to close this, please reopen (and help us reproduce it) if it's still a problem

closed 2017-02-07 by jackkoenig