chipsalliance/chisel #350: Wrong Gender Errors?

donggyukim, 2016-11-03:
224ee397db324067b35874bf22c76a63e9ca531b breaks BOOM compliation in FIRRTL generating wrong gender errors as follows:

firrtl.passes.PassExceptions:
firrtl.passes.CheckGenders$WrongGender:  @[execute.scala 659:29]: [module ALUMemExeUnit]  Expression lsu.io.dec_st_vals is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[execute.scala 660:29]: [module ALUMemExeUnit]  Expression lsu.io.dec_ld_vals is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[execute.scala 663:29]: [module ALUMemExeUnit]  Expression lsu.io.commit_store_mask is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[execute.scala 664:29]: [module ALUMemExeUnit]  Expression lsu.io.commit_load_mask is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 629:32]: [module RenameStage]  Expression bsy_table.io.p_rs[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 630:32]: [module RenameStage]  Expression bsy_table.io.p_rs[1] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 637:35]: [module RenameStage]  Expression bsy_table.io.p_rs[2] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 670:39]: [module RenameStage]  Expression freelist.io.req_preg_vals[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 680:37]: [module RenameStage]  Expression freelist.io.enq_vals[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 683:37]: [module RenameStage]  Expression freelist.io.enq_pregs[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 685:37]: [module RenameStage]  Expression freelist.io.ren_br_vals[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 686:37]: [module RenameStage]  Expression freelist.io.ren_br_tags[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 690:40]: [module RenameStage]  Expression freelist.io.rollback_wens[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 693:40]: [module RenameStage]  Expression freelist.io.rollback_pdsts[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[rename.scala 695:37]: [module RenameStage]  Expression freelist.io.com_wens[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 358:40]: [module BOOMCore]  Expression dec_brmask_logic.io.is_branch[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 359:40]: [module BOOMCore]  Expression dec_brmask_logic.io.will_fire[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 393:41]: [module BOOMCore]  Expression rename_stage.io.dis_inst_can_proceed is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 417:44]: [module BOOMCore]  Expression rename_stage.io.wb_valids[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 420:44]: [module BOOMCore]  Expression rename_stage.io.wb_pdsts[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 429:44]: [module BOOMCore]  Expression rename_stage.io.wb_valids[1] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 434:44]: [module BOOMCore]  Expression rename_stage.io.wb_pdsts[1] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 429:44]: [module BOOMCore]  Expression rename_stage.io.wb_valids[2] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 434:44]: [module BOOMCore]  Expression rename_stage.io.wb_pdsts[2] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 441:31]: [module BOOMCore]  Expression rename_stage.io.com_valids is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 443:35]: [module BOOMCore]  Expression rename_stage.io.com_rbk_valids is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 499:29]: [module BOOMCore]  Expression issue_unit.io.dis_valids is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 510:33]: [module BOOMCore]  Expression issue_unit.io.fu_types[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 558:38]: [module BOOMCore]  Expression register_read.io.iss_valids[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 642:29]: [module BOOMCore]  Expression ALUMemExeUnit_1.io.lsu_io.dec_st_vals[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 644:29]: [module BOOMCore]  Expression ALUMemExeUnit_1.io.lsu_io.dec_ld_vals[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 650:29]: [module BOOMCore]  Expression ALUMemExeUnit_1.io.lsu_io.commit_store_mask is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 651:29]: [module BOOMCore]  Expression ALUMemExeUnit_1.io.lsu_io.commit_load_mask is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 730:22]: [module BOOMCore]  Expression rob.io.dis_valids is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 750:38]: [module BOOMCore]  Expression rob.io.debug_wb_valids[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 767:43]: [module BOOMCore]  Expression rob.io.debug_wb_wdata[0] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 750:38]: [module BOOMCore]  Expression rob.io.debug_wb_valids[1] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 773:43]: [module BOOMCore]  Expression rob.io.debug_wb_wdata[1] is used as a FEMALE but can only be used as a MALE.
firrtl.passes.PassException: 38 errors detected!

aswaterman, 2016-11-03:
If you use the compatibility layer, you might get Firrtl errors.  It disables the gsdt checking, because the gsdt checking introduces backwards incompatibilities.

It looks like only a small number of lines trigger the issue.  We should better understand the problem, because it might not actually be a chisel problem.

donggyukim, 2016-11-03:
I could compile BOOM with the previous commits, so I guess it may generate wrong CHIRRTL. @ccelio has some ideas?

aswaterman, 2016-11-03:
Does the exact same input compile with Chisel before the gsdt branch was merged?

Either way, we might need to admit the the compatibility layer is never going to fully work.  This commit was necessary to fix backwards incompatibilities in main-line rocket-chip: for example, it used to be legal to write Wire(foo) where now you'd have to write Wire(foo.flip).  Unless there truly is a bug in this commit, it might be better just to change BOOM.

ccelio, 2016-11-03:
I don't understand any of FIRRTL's gender nonsense.  Although I only did a preliminary scan of the code, it looks like it's freaking out about `Vec(n, Something(INPUT))` signals being on the left-hand side. Whisky Tango Foxtrot.

````
firrtl.passes.CheckGenders$WrongGender: @[core.scala 441:31]: [module BOOMCore] Expression rename_stage.io.com_valids is used as a FEMALE but can only be used as a MALE.
````

Like that one should be incredibly straight-forward to examine:

````
//rename stage IO
val com_valids = Vec(pl_width, Bool(INPUT))

//ROB IO
val com_valids  = Vec(machine_width, Bool(OUTPUT))

// core connection
rename_stage.io.com_valids := rob.io.commit.valids
````

aswaterman, 2016-11-03:
Why are you declaring internal signals with directions?

aswaterman, 2016-11-03:
Oh, I see what you're describing, n/m.

ccelio, 2016-11-03:
Andrew, you nearly gave me a heart attack (my response was going to be "because it's worked for 5 years?").

I think there are two issues here:

1) BOOM should be updated to whatever the current Chisel3 strict standards are.

2) The error message shouldn't be brain damaged.

aswaterman, 2016-11-03:
In the mean time, you might try e.g.

    Vec(pl_width, Bool()).asInput

Moving to strict chisel3 isn't going to be all that straightforward, since some of the changes can't even be detected statically (e.g., UInt(1) is a width for some retardo reason).

ccelio, 2016-11-03:
> (e.g., UInt(1) is a width for some retardo reason).

Why the fuck is that allowed?

aswaterman, 2016-11-03:
You should ask @jackbackrack et al, who just released a beta version that does this against my repeated warnings.  It's a terribly dumb idea that will confuse anyone who has been using chisel.  I think they're kowtowing to Google for god knows what reason.

aswaterman, 2016-11-03:
I'd be fine if they just deleted the constructor, but repurposing it in a syntactically indistinguishable way is the screwup.

ccelio, 2016-11-03:
Agreed, nuke it so nobody can use it if need be; but changing five years of code to mean something even less obvious is dumb, dumb, dumb.

aswaterman, 2016-11-03:
You should make your opinions known at chisel meetings.

sdtwigg, 2016-11-03:
The ship has basically sailed on "code written 5 years ago still works" with chisel3. Particularly since the original claim was that the only backwards incompatibility was that you would need to wrap wires in Wire().

We can levy the same argument against every backwards incompatible change that was foisted upon the language on the conversion. The key is to have answers that improves the UX, not compiler implementation. (Also, I do agree with parts of these changes, even though as a whole they are a mess.)
First, quickly going down the stated list of the changes:
* Why must Vec(Reg) be replaced with Vec(Reg) and what about the cases where the registers were variably initialized? (I'm actually going to touch on this further since this hints at but in no way explains the drastic Vec semantics changes in chisel3.)
* What is the benefit to the user of wrapping everything in Wire? Beforehand you were substantially less likely to get an undefined reference error but standard chisel3 will just vomit them at you (occasionally archaically).
* Why is there a rename from clone to cloneType? (I am actually curious about this one since you can't easily detect this statically either due to the default cloneType implementation)
* What does "bulk-connection to a node that has been procedurally assigned-to is illegal" mean? (in particular, 'procedurally' and asserting illegality seems to contradict the last bullet in the second list.)
* Why where the arguments to Mem and Vec swapped?
* Why can't I do masked writes to more general things like before?
* Why are connections between SInt and UInt illegal?
* Why were <> and := just completely rewritten? The migration guide only hints at the changes when really they exhibit drastically different behaviors in anything but the simple cases (particularly when aggregates are involved). (This is the one change that I feel was made with minimal to negative merit and benefit.)

Second, considering the undocumented changes:
* Inspired by the actual posted issue, why is gender now a thing in Chisel and why does it show up so prominently in error messages?
* For Bundle definitions, why was Vec.fill 'broken'? (This is still not mentioned in the migration guide since, apparently, it is rare to have a Vec in a Bundle?)
* Why is <> now affected by when statements? (Even generous readings of the porting guide don't indicate this is now true.)
* Where did addPin go?
* What was the benefit of changing the semantics of Vec(Seq(...)) so drastically? You can see from #336, #348, #326, #286, #241 that the new semantics of Vec are just unclear. Perplexingly, the documentation is wrong on multiple counts:

> Creates a new [[Vec]] composed of elements of the input Seq of [[Data]] nodes.

This described what the old chisel2 Vec does. Chisel3 creates a Wire of Vec of the input data type. Then, it 'connects' the inputs to that new Wire; however, this does require the user understand exactly what it means in chisel3 to 'connect' something. This occasionally involves a conversation with the user on how an internal node marked as an input actually means flipped which reverse 'some' connection statements (aggregate versus element). Regardless, the underlying distinction of putting the things in a dynamically indexable container versus connecting them to a Wire(Vec(...)) causes very different behaviors in practice.

> input elements should be of the same type (this is checked at the FIRRTL level, but not at the Scala / Chisel level)

This misunderstands types because, pedantically, all Data is the same type: Data. Also, it is possible to construct various (actually reasonable) examples where things of the concrete type do not work:
* Putting in a Seq of all UInt with varied directionality (including no-direction) and then doing a dynamically indexed read will not be able to successfully read all values.
* You can actually use a Seq of mixed Bool and UInt (as long as a Bool isn't the first element!).

> the width of all output elements is the width of the largest input element
This isn't true at all when given Aggregate types of varying width.

>output elements are connected from the input elements
This is actually somewhat accurate. However, it contradicts the actual summary which says 'composed' and not connected. (Also, if the output is merely connected to the input, then what IS the output? Answer: Wire, but never stated).

At least with "UInt(_: Int)", you can make a reasonable argument to software engineering practices of not having different variants of the same overloaded function do and return different things. Note the Vec.apply function is similarly egregious. However, at least in Chisel2 (partly due to not needing Wire), the returned Vec from any of them could generally be used in the same way. However, as discussed above, in Chisel3, the Vec(Seq(...)) is substantially different.

ccelio, 2016-11-09:
@aswaterman Sorry for not following up on your suggestion `Vec(pl_width, Bool()).asInput` earlier. That does appear to be the solution.

So... do we close this issue and just hope everybody knows to convert over their Vec IOs? Or is this an issue that can be caught/managed by Chisel/FIRRTL? The current error message leaves a lot to be desired.

aswaterman, 2016-11-11:
Glad it worked.

I think the right way to get better error messages is to write chisel3 code instead.  So I think we should close this as "known workaround, won't fix."

closed 2016-11-11 by aswaterman