chipsalliance/chisel #352: Cannot use vector of registers in sequential mux circuit

edwardcwang, 2016-11-07:
I get a "Expression key already in stack" error using Vec.fill(n) or Seq.fill(n) or "Not bound to synthesizable node, currently only Type description" error with Vec(Reg()) construct. Code is at the bottom of this issue. @chick

With `val num_ones = Vec(Seq.fill(n)(Reg(init=UInt(0, width = 32))))` or `val num_ones = Vec.fill(n){Reg(init=UInt(0, width = 32))}`

[info]   firrtl_interpreter.InterpreterException: Expression key num_ones_0 already in stack
[info]   at firrtl_interpreter.ExpressionExecutionStack$$anonfun$push$1.apply(ExpressionExecutionStack.scala:42)
[info]   at firrtl_interpreter.ExpressionExecutionStack$$anonfun$push$1.apply(ExpressionExecutionStack.scala:36)
[info]   at scala.Option.foreach(Option.scala:257)
[info]   at firrtl_interpreter.ExpressionExecutionStack.push(ExpressionExecutionStack.scala:36)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:273)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$4.apply(LoFirrtlExpressionEvaluator.scala:429)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$4.apply(LoFirrtlExpressionEvaluator.scala:424)
[info]   at firrtl_interpreter.Timer.apply(Timer.scala:40)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.firrtl_interpreter$LoFirrtlExpressionEvaluator$$resolveDependency(LoFirrtlExpressionEvaluator.scala:423)
[info]   at firrtl_interpreter.LoFirrtlExpressionEvaluator.getValue(LoFirrtlExpressionEvaluator.scala:81)

With: `val num_ones = Vec(n, Reg(init=UInt(0, width = 32)))`

[info]   chisel3.core.Binding$BindingException: 'this' (chisel3.core.UInt@28): Not bound to synthesizable node, currently only Type description
[info]   at chisel3.core.Binding$.checkSynthesizable(Binding.scala:185)
[info]   at chisel3.core.Bits.binop(Bits.scala:159)
[info]   at chisel3.core.UInt.do_$plus$amp(Bits.scala:438)
[info]   at chisel3.core.UInt.do_$plus$percent(Bits.scala:440)
[info]   at chisel3.core.UInt.do_$plus(Bits.scala:420)
[info]   at utils.Debouncer$$anonfun$1$$anonfun$apply$mcV$sp$1.apply$mcVI$sp(Debouncer.scala:46)
[info]   at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:166)
[info]   at utils.Debouncer$$anonfun$1.apply$mcV$sp(Debouncer.scala:45)
[info]   at chisel3.core.WhenContext.<init>(When.scala:57)
[info]   at chisel3.core.when$.apply(When.scala:31)

Debouncer.scala: 

DebouncerUnitTest.scala:

seldridge, 2016-11-07:
Try the following: 
```scala
val num_ones = Reg(Vec(n, UInt(width=32)), init = Vec(Seq.fill(n)(UInt(0))))
```
Putting a `Reg` inside a `Vec` is deprecated in my understanding.

sdtwigg, 2016-11-07:
Implementation-wise, this should also work:
val num_ones = Reg(init = Vec(n, UInt(width=32)).fromBits(0.U))
or
val num_ones = Reg(init = Vec.fill(n)(UInt(0, width=32)))

chick, 2016-11-07:
Yes, I agree the idioms you suggest
Implementation-wise, this should also work:
```
val num_ones = Reg(init = Vec(n, UInt(width=32)).fromBits(0.U))
//or
val num_ones = Reg(init = Vec.fill(n)(UInt(0, width=32)))
```
Are better.  Are you able to run your circuit correctly now

chick, 2016-11-08:
@edwardcwang , I'd just like to confirm that you are able to progress with Debouncer.  If so @ucbjrl and I will use this PR to try and flag the wrong use of Vec of Reg, in an upcoming version

edwardcwang, 2016-11-08:
@chick @ucbjrl Yes, I can confirm that all three of the following variants that put Vec inside Reg don't have the error.

```
val num_ones = Reg(init = Vec(n, UInt(width=32)).fromBits(0.U)) // nominally preferred
val num_ones = Reg(init = Vec.fill(n)(UInt(0, width=32)))
val num_ones = Reg(Vec(n, UInt(width=32)), init = Vec(Seq.fill(n)(UInt(0))))
```

jackkoenig, 2016-11-15:
This looks resolved?

closed 2016-11-15 by jackkoenig

edwardcwang, 2016-11-15:
@jackkoenig Any chance that this could get flagged as an error to help future users?

jackkoenig, 2016-11-15:
Probably, I'll see what I can do