chipsalliance/chisel #353: HelloWorld example is missing from chisel3 README and website

ccelio, 2016-11-07:
Per this stackoverflow question:

http://stackoverflow.com/questions/40470153/is-there-a-simple-example-of-how-to-generate-verilog-from-chisel3-module

How do we create a simple Chisel3 HelloWorld? The old Chisel2 README and website provided a bare bones `build.sbt` and `hello.scala` that could get Verilog from Chisel. 

It's not obvious how to do the same thing in Chisel3 and I believe the README and website should cover this.

-------
The old way:

````
$ cat build.sbt
scalaVersion := "2.11.7"

libraryDependencies += "edu.berkeley.cs" %% "chisel" % "latest.release"
````
````
$ cat Hello.scala
import Chisel._

class HelloModule extends Module {
  val io = new Bundle {}
  printf("Hello World!\n")
}

class HelloModuleTests(c: HelloModule) extends Tester(c) {
  step(1)
}

object hello {
  def main(args: Array[String]): Unit = {
    chiselMainTest(Array[String]("--backend", "c", "--compile", "--test", "--genHarness"),
       () => Module(new HelloModule())){c => new HelloModuleTests(c)}
  }
}
````

ucbjrl, 2016-11-07:
We should direct people to the chisel-template repo (and wiki).

ccelio, 2016-11-07:
Yah, that's definitely a great resource.

But is it possible though to cut down a hello-world to just two or three files in a flat hierarchy, like the previous chisel2 example?  Being only ~15 lines of code in one directory across two files is really, really nice and much less intimidating.

ucbjrl, 2016-11-07:
We want people to start with something that can evolve into a complete Chisel(3) project, and the consensus is that starting off in a flat hierarchy is not a good idea. The goal is to have people think about testing from the very beginning, and structure their projects to reflect this.

Although sbt and most IDEs can work with a non-standard layout for trivial projects, once the project becomes complex enough such that the flat hierarchy starts introducing problems, significant work may be required to correct it.

Also, it's easier to diagnose problems if we adhere to a standard structure for Chisel projects.

ccelio, 2016-11-07:
Thanks to @jackkoenig , the solution for a minimal chisel3 example is:

> build.sbt in the root directory:
```
scalaVersion := "2.11.8"
resolvers ++= Seq(
  Resolver.sonatypeRepo("snapshots"),
  Resolver.sonatypeRepo("releases")
)
libraryDependencies += "edu.berkeley.cs" %% "chisel3" % "3.0-SNAPSHOT"
```
> And adding to the source codel:
```
object GCDDriver extends App {
  chisel3.Driver.execute(Array.empty[String], () => new GCD)
}
```

> You can then generate the Verilog by running:  sbt "run-main GCDDriver". 

----

I think that's so small and beautiful it deserves to be front and center.  I think it really lowers the bar for people to dip their toes into Chisel3. And then right next to it, a link to the project chisel-template.

chick, 2016-11-07:
I like Jack's solution to but with one caveat.  The following is better
```
object GCDDriver extends App {
  chisel3.Driver.execute(args, () => new GCD)
}
```
This way you can allow the user to change options easily.  You can see those
options with
```
bash> sbt 'run-main GCDDriver --help'
```

jackkoenig, 2016-11-07:
Yeah that is better, I'll modify my answer on Stack Overflow

schoeberl, 2016-11-09:
Ok, but where does GCD come from? I do not assume it is part of Chisel 3.

schoeberl, 2016-11-09:
Ok, see now it is coming from an example repository. However, I REALLY want to see a simple setup that is described in a README.md with two or three files, similar to the Chisel 2 setup. Just explaining - no need to do a git clone. Who would start a small C/Java/Scala/Python/VHDL/... project by cloning a git repository? I think git should even be optional when wanting to use Chisel.

Ok, ok, stop ranting ;-)

Martin

ucbjrl, 2016-11-09:
Some of us feel that it is important to start every project from a layout that can evolve. The fear is that people will start with a "simple" setup, then encounter problems once their design grows beyond a single file or two - problems that wouldn't arise if they had started from a "standard" layout.

Others agree with you that there is a need for a "simple" setup to get people excited about Chisel.

The discussion is ongoing.

schoeberl, 2016-11-09:
I completely understand. A well structured layout of a project is very important. And when Scala/sbt gives some usage patterns (e.g., convention over configuration) I am very much in it. However, as a starting point a “Hello World’ should be VERY minimalistic. I assume people know that this “Hello World” is not the starting point of a decent sized project. At least I hope so ;-)

Martin

ducky64, 2017-02-06:
Resolution: write documentation on how to get started really quickly with Chisel (i.e. minimal build.sbt, Top wrapper, and main file), along with more documentation on how to structure Chisel projects properly.

https://stackoverflow.com/questions/40470153/is-there-a-simple-example-of-how-to-generate-verilog-from-chisel3-module

ducky64, 2017-11-08:
Resolution: add this to readme and close

ekiwi, 2019-10-18:
This seems to be fixed now: https://github.com/freechipsproject/chisel3#what-does-chisel-code-look-like

closed 2019-10-18 by ekiwi