chipsalliance/chisel #356: Windows

schoeberl, 2016-11-09:
I really think you need to have some support for Windows (cygwin might be ok) if you want to go mainstream. We might often overlook it, but the user base, also in EDA, is huge.

Martin

ducky64, 2016-12-16:
Chisel itself should work on Windows (it's Scala, which is mostly platform agnostic). The main problem is Verilator, which doesn't have Windows binaries (and "here's the 20 steps to build Verilator on your Windows machine" doesn't sound too attractive, either). That being said, if someone wants to release Verliator binaries for Windows, making the rest of Chisel work with it shouldn't be too difficult.

There's been ongoing work on a FIRRTL interpreter written in Scala, which for most simple designs could replace Verilator.

donggyukim, 2016-12-16:
I'm running both Chisel and Verilator on Windows with Cygwin. I installed JDK and SBT to run Chisel and some Cygwin packages to compile Verilator. It's also known that Verilator can be compiled with Visual Studio.

@schoeberl I believe we can close this issue?

ducky64, 2016-12-16:
I don't think the issue should be closed until the install process is documented.

Verilator itself does not compile under MSVC, though it claims its output does.

I tried building Verilator under MSYS/MinGW/GnuWin32, and it turned out to be pretty painful without much results. Ideally what we (or someone) would be able to do is compile Verilator under Windows and release binaries that run natively, without Cygwin (as the rest of Chisel would).

ducky64, 2017-02-06:
Seems like this is working with Cygwin, per @donggyukim 
If anyone wants to provide a native Windows build of Verilator, we welcome your PR to update Chisel documentation.

closed 2017-02-06 by ducky64

edwardcwang, 2017-02-06:
Maybe we should also document this? The README still has an effectively blank Windows section...

schoeberl, 2020-02-17:
Just for completeness: Chisel works fine under Windows native without Cygwin: for plain Chisel (using the tester) and Verilog generation (no Verilator needed). Just install sbt, OpenJDK8, and IntelliJ.