chipsalliance/chisel #361: chisel3 exception handling

ucbjrl, 2016-11-10:
We should have a top-level exception handler in chisel3. It should report accumulated errors and warnings before exiting.

aswaterman, 2016-11-11:
To the extent that we can continue executing past errors, so that we can report as many errors as possible without exiting, I agree.  But in the unresumable case, the stack trace is helpful, so the top-level exception handler should still print it.

ducky64, 2016-12-15:
So once an exception is thrown and it reaches the top level handler, can you actually resume from where the exception was thrown?
(under the assumption that the exception is non-fatal to the current pass, and where the possibility for more similar exceptions exist)

ducky64, 2017-02-06:
Resolution: place a "top-level" exception handler in (or around) Driver.elaborate, which just prints out the error (with a stack track) along with other non-fatal errors.

ucbjrl, 2019-01-10:
To do this correctly will require significant restructuring of the code in order to avoid reporting the same error multiple times. I'm submitting a PR that demonstrates a simple (naive) solution, but marking it DO NOT MERGE. We should discuss this at a development meeting, but for the moment I'm bumping the milestone.

jackkoenig, 2022-03-22:
This is resolved via Stage/Phase in Chisel 3.3.

closed 2022-03-22 by jackkoenig