chipsalliance/chisel #363: Abolish Module() Wrapping

jackkoenig, 2016-11-11:
While implementing #362 I looked in to how we might get ride of Module() wrapping. Many of the original Chisel people probably know much of the following, but it's new to me.

For others, the reason why you have to wrap Module instantiations in Module() is because we need the ability to execute some code after running the constructor of a given Module. It turns out, Scala *does* provide a way to do this:  [DelayedInit](http://www.scala-lang.org/api/current/scala/DelayedInit.html). Unfortunately, it was deprecated in 2.11.0.

Now, this discussion is less relevant with the error detection in #362 since forgetting to wrap with Module will become virtually a non-problem. However, we could remove the need for Module() wrapping by using DelayedInit. Despite deprecation, [DelayedInit will not be removed until a suitable replacement is found](https://github.com/scala/scala/releases/tag/v2.11.0-RC1). There is a [proposed replacement](https://issues.scala-lang.org/browse/SI-4330?jql=labels%20%3D%20delayedinit%20AND%20resolution%20%3D%20unresolved), but it is still open. Is it possible to suppress a single deprecation warning?

aswaterman, 2016-11-11:
We tried, but failed, to use DelayedInit to solve this problem back when we did Chisel2.

Good news is that if the proposed replacement eventually works, we can turn Module.apply into a no-op, so that new code can omit it but old code still works.

jackkoenig, 2016-11-11:
Do you remember why it didn't work?

The only problem I see in Chisel3 is that you need to be able to set the currentModule back to the parent which is done in Module.apply by putting it in a local variable. This is effectively using the call-stack as a moduleStack as originally proposed in #362.

Am I missing something else?

aswaterman, 2016-11-11:
I believe @jackbackrack investigated it, but IIRC the non-trivial examples caused null to be dereferenced.  Of course, that is not proof that it cannot work.  You could try it.

jackkoenig, 2016-11-12:
It appears DelayedInit has a huge problem. If you create a class with no body, then DelayedInit doesn't get called.  https://issues.scala-lang.org/browse/SI-4680.

ducky64, 2017-02-06:
Not happening anytime soon, unless there is some kind of post init hook in Scala.

closed 2017-02-06 by ducky64