chipsalliance/chisel #368: No errors when putting literal in IO

ducky64, 2016-11-16:
This code is obviously wrong but compiles and elaborates without errors:
```
import chisel3._

class CRModule extends Module {
  val io = IO(new Bundle {
    val a = Output(32.U)
  })
  io.a := 64.U
}
```

I think what it ends up doing is discarding the literal and only taking the type and bitwidth.

The more insidious version is where you use the UInt(width=...) constructor but forget the width= key. One solution may be to deprecate the UInt(...) constructor, and possibly also the UInt(width=...) constructor. Related: #294

Whether there should be some kind of truncation warning is another question.

sdtwigg, 2016-11-16:
In current chisel3, Output, like Input, Flipped, Reg, Vec (some), Wire, etc. immediately clones its given argument in order to just get an unbound type without checking whether said argument had actual hardware meaning attached to it. 

You would get the same result for more absurd things like:
```scala
val a = Output(Reg(init=0.U))
val a = Output(Reg(UInt(width=16))) // identical to dropping Reg
val a = Output(Vec.fill(4)(1.U))
```

#294 isn't related to the root issue. It namely affects the issue that these two statements look similar but act differently. (Note, behavior shifts depending on compatibility layers.)
```scala
val a = Output(UInt(16)) // in most versions, creates literal 16 then clones to UInt()
val a = Output(UInt(width=16)) // good
```

aswaterman, 2016-11-16:
This is not a bug

ducky64, 2016-11-16:
You don't want stricter checking and warnings when Chisel (silently) discards data the user specified?

aswaterman, 2016-11-16:
It's a common pattern to write things like Reg(value) or Output(value) to make a Reg or Output that has the same type as value.  This is just an (unfortunate) instance of that pattern.

Why stop at literals?  Why not ban the pattern in general and make the language exceedingly difficult to use to write generators, in the name of usability for trivial examples?

ducky64, 2016-11-16:
This seems to be another trade-off between safety and conciseness, the more precisely you specify something, the higher the chance of the compiler catching mistakes before you spend hours debugging (or worse, don't find the bug at all). It's like the trade-off between using Scala and Python: dynamic typing means there's less effort involved in writing Python, but it's harder to guarantee your code is correct.

Anyways, the solution here doesn't have to be "ban the pattern", but it still seems counter-intuitive that the compiler is silently dropping data the user put in. Perhaps a compromise solution, like strict checking requiring a "cast-to-type" operation. For example, something like:
```
val value = Reg(UInt(init=16.U))
...
val a = Output(value.asType)
```
When you assign a literal to an IO, someone might (reasonably) expect that the value of the IO is the literal. This would allow the compiler to catch those potential issues and ensure that it's obvious any value is being dropped. That the conversion is explicit instead of implicit also vastly decreases the chances of a mistake.

sdtwigg, 2016-11-16:
I meant to include in my original response (which I was trying to keep simply factual) that I do not consider this a bug since all the functions are working as specified. It is more a consequence of various API parts interacting in an unfortunate manner.

Since the following comments delve more into the bug vs. feature debate: As a matter of opinion, I feel this falls in the general class of Chisel 'misfeatures' that purports to make things easier but just ends up setting annoying-to-debug traps for users.

aswaterman, 2016-11-16:
Values-as-types does fall into the camp of unintuitive features.  OTOH, Chisel2 disallowed it in some contexts, and the effect was to create a different set of pitfalls (and a proliferation of explicit cloneType calls).

azidar, 2016-12-02:
But if we provide an explicit .asType (or I'd preferred a .getType method), then it does make it clear that a type is different than a value, which is something we want to emphasize to new users. My guess is people used cloneType because a different API didn't exist. We can easily port between APIs by adding a deprecation warning instead of an error in the Chisel3 compatibility mode, and then have an error in chisel3._.

@aswaterman, do you have examples where replacing Reg(value) with Reg(value.getType) wouldn't work?

aswaterman, 2016-12-02:
@azidar if getType does what I'd expect (return a node with the same type but without an associated value), then I would think that replacing Reg(value) with Reg(value.getType) would always work.

However, I'd still advise against changing anything here.  The use case isn't for things like Reg(1.U); it's for things like Reg(some.data.structure), where you just want to create a Reg with the same type as something, but you don't explicitly know its type.  The getType thing adds boilerplate for what in my opinion is a dubious improvement in clarity.

azidar, 2016-12-02:
I think for a new user, Reg(value) would mean the register's initially set to value, or it could mean you are delaying value by one cycle. While I agree for expert users, this is a dubious improvement in clarity, for new users this change would really help clarify what the API actually is.

ccelio, 2016-12-02:
I believe we should require `getType` (or `.clone` or whatever); the current behavior leads to ambiguous and unreadable code, IMO.  

I'm skimming through rocket's code, and this scenario is very rare (there are 320 `Reg` uses and I'm seeing roughly 6 uses of `Reg(thing_with_values)`).

Let me give an example:

````scala
class MulDiv(...) extends Module {                               
  val io = new MultiplierIO(...)                                                           
  ...                                                                                                                                                                       
  val req = Reg(io.req.bits)                                                                               
````

To me, that looks like we are registering `io.req.bits` and passing it to `val req`.  But it's not -- it's actually only creating a Register of the same type.  Only much later does it perform a `req := io.req.bits`.

My concern here is other people who aren't the original author reading code with `Reg(io.req.bits)` or `Wire(io.req.bits)` will be unnecessarily misled. 

-----

P.S.: I'm only supporting `getType` assuming it doesn't show up in a lot of false positives and add a ton of code verbosity.

sdtwigg, 2016-12-03:
Users, particularly new ones, are far too prone to making this mistake. So, I have found making "passing a bound hardware node to somewhere expecting a raw type" an error to drastically reduce the time spent in tedious debugging sessions. (Note that, because Chisel in general allows you to have undriven wires and uninitialized registers without warnings or Xs, it can take a while to find where the root error is).

Plus, approach this as someone who is reading someone else's code that needs changing:
Seeing the 'newType' (what I called it in initial PR)/'chiselCloneType' (what is what renamed to)/'getType' (what chiselCloneType is presumably being aliased as) assures you that they meant for the value discard to happen.

For example, if I read Chris's example 'Reg(io.req.bits)', I know that it is discarding the value but have no idea which of the following the original writer truly wanted:
1. Reg(io.bits.newType)
2. Reg(init = io.bits) or  Reg(t = io.bits.newType, init = io.bits) (See *)
3. Reg(next= io.bits) or Reg(t = io.bits.newType, next = io.bits)

* The difference during width inference between these two (which only exists when io.bits is precisely a UInt or SInt but not an Aggregate is especially 'fun')

closed 2017-07-31 by ducky64