chipsalliance/chisel #375: Should there be an error message on Vec(Bits)?

ccelio, 2016-11-19:
I ran into a problem in my code, and I'm wondering if Chisel (or FIRRTL) should catch this error:

````scala
val myVec = Vec(8, Bool())
val myBitVector = Reg(UInt(width=8))
myVec := Vec(myBitVector)
````

The full value of `myBitVector` is being assigned into `myVec[0]` by Chisel, and the higher 7 elements are randomized.

To get the intended behavior, I should have done this:

````scala
myVec := Vec(myBitVector.toBools)
````

-------

Should ".toBools" be required? Should Vec() error out because of the incompatibility of MyBitVector and myVec[0]? Is the current behavior ever useful?

jackkoenig, 2016-12-06:
You actually could just write:
```scala
myVec := myBitVector.toBools
```

Since toBools returns a Seq and Vecs can be connected from Seqs. *(also, don't forget to wrap myVec in Wire)*

The `Vec.apply` method called by `Vec(Bits)` is intended to be analogous to `Seq.apply` for constructing Seqs with initial values, eg:

```scala
val xs = Seq(1, 2, 3)
```

It's useful for constructing Vecs from things, eg.
```scala
val myConsts = Vec(0.U, 1.U, 2.U, 3.U)

val reg1 = Reg(init = 1.U)
val reg2 = Reg(init = 2.U)
val myRegs = Vec(reg1, reg2)
```

jackkoenig, 2016-12-06:
I haven't thought this through yet (or looked so see if it's used), but it might be prudent in chisel3._ to bring Vec connection inline with Bundle in strictness in that Vecs being connected must be of the same size

aswaterman, 2016-12-06:
Given that it isn't especially difficult to write things like

    for ((e1, e2) <- v1 zip v2) e1 := e2

and

    smallVec := bigVec.slice(0, smallVec.size)

I think that favoring type safety here is worth the extra verbosity in the rare case that the user intends to connect Vecs of differing size.

jackkoenig, 2016-12-06:
It looks like this should already be an error in chisel3, investigating...

ccelio, 2016-12-06:
I'm importing `Chisel._` if it makes a difference.

jackkoenig, 2016-12-06:
I thought I was importing chisel3._ but it looks like I was doing Chisel._ as well. So I'm guessing we don't really need to do anything then? It's an error in chisel3._, but we probably shouldn't change it in the compatibility wrapper.

ccelio, 2016-12-06:
Got it. I just suck.

aswaterman, 2016-12-06:
Yeah, taking no action at all seems like the right course of action.

closed 2016-12-06 by jackkoenig