chipsalliance/chisel #381: Issues with <> in chisel3._

jackkoenig, 2016-11-22:
I might just be doing something really stupid so if that's the case please let me know.

In chisel3._, when I attempt to do a simple bulk connect with DecoupledIOs:
```scala
class BulkConnectTop extends Module {
  val io = IO(new Bundle {
    val in = Input(Decoupled(UInt(64.W)))
    val out = Output(Decoupled(UInt(64.W)))
  })

  io.out <> io.in
}
```
I get the following invalid Firrtl:
```
circuit BulkConnectTop :
  module BulkConnectTop :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip in : {flip ready : UInt<1>, valid : UInt<1>, bits : UInt<64>}, out : {flip ready : UInt<1>, valid : UInt<1>, bits : UInt<64>}}

    io is invalid
    io is invalid
    io.out.bits <= io.in.bits @[BulkConnect.scala 15:10]
    io.out.valid <= io.in.valid @[BulkConnect.scala 15:10]
    io.out.ready <= io.in.ready @[BulkConnect.scala 15:10]
```

This errors in Firrtl, and there are two invalidations of io for some reason.

The same code code using Chisel._ works as expected:
```scala
class BulkConnectTop extends Module {
  val io = IO(new Bundle {
    val in = Decoupled(UInt(64.W)).asInput                                                                                                                                                                    
    val out = Decoupled(UInt(64.W))
  })

  io.out <> io.in
}
```
(Granted, I'm a little confused by .W working? I guess it works but just won't ever be required in compatibility mode?)
It gives perfectly sensible Firrtl:
```firrtl
circuit BulkConnectTop :
  module BulkConnectTop :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip in : {flip ready : UInt<1>, valid : UInt<1>, bits : UInt<64>}, out : {flip ready : UInt<1>, valid : UInt<1>, bits : UInt<64>}}

    io is invalid
    io is invalid
    io.out <- io.in @[BulkConnect.scala 15:10]
```

This Firrtl compiles just fine to Verilog

jackkoenig, 2016-11-22:
Yep, dumb mistake. `Input(...)` and `Output(...)` coerce the ports to be all Input or Output. Since `DecoupledIO` already gives direction, you need to use `Flipped` (or oldschool `.flip()`)

```scala
class BulkConnectTop extends Module {
  val io = IO(new Bundle {
    val in = Flipped(Decoupled(UInt(64.W)))
    val out = Decoupled(UInt(64.W))
  })

  io.out <> io.in
}
```

closed 2016-11-22 by jackkoenig

sdtwigg, 2016-11-22:
The original example should have been coercing all of the directions of in and out (you shouldn't get a FIRRTL error from this). However, I am no longer confident in the desired behaviors of certain constructs like this in certain packages so this error may be intended behavior.