chipsalliance/chisel #388: Weird Width Decisions

jackkoenig, 2016-11-30:


aswaterman, 2016-12-01:
......oops

jackkoenig, 2016-12-01:
Related: "h0a".U should have a width of 4 right? "h1a".U should have a width of 5?

aswaterman, 2016-12-01:
5 is probably the best value. I would just handle this by parsing as a BigInt and calling bitLength.

jackkoenig, 2016-12-01:
That was my plan, what about for signed though? .bitLength + 1?

jackkoenig, 2016-12-01:
Just kidding, can't do .S on Strings

mwachs5, 2016-12-01:
Why is 5 better for 0a than 8?

aswaterman, 2016-12-01:
I think either is justifiable. My argument for the smaller value is that
there's no difference between writing 0xa.U and 0x0a.U, so "a".U and "0a".U
should also be the same as each other. But I could be convinced otherwise.

mwachs5, 2016-12-01:
Sure, I agree with that. I thought you were saying 0a could be inferred to
width 5.

aswaterman, 2016-12-01:
My bad, I meant 0a should be 4 bits and 1a should be 5 bits

commit 86f15d92 (2016-12-01): Fix literal width

Fixes #388
PR #389 (2016-12-05): Fix literal width
Fixes #388 

Also add sanity checks
commit a1e8ae80 (2016-12-05): Fix literal width (#389)

Fixes #388
closed 2016-12-05 by jackkoenig