chipsalliance/chisel #399: Unhelpful Subword Assignment Error (in Firrtl)

jackkoenig, 2016-12-03:
See http://stackoverflow.com/questions/40950073/chisel-3-assignment-to-bit-range

```scala
class TestX extends Module
{
    val io = IO(new Bundle {
        val a = Output(UInt(width=2))
    })

    io.a(1, 0) := UInt(0)
}
```

Gives error:
`Error: [module TestX] Expression T_4 is used as a FEMALE but can only be used as a MALE.`

We should probably try to catch this in Chisel

jackkoenig, 2016-12-03:
Also we should probably document this limitation

ducky64, 2017-02-06:
Subword assignments not supported.

It may be possible to emit a better error by adding a SourceOnlyBinding or something, so assigning to it in Chisel frontend will give the user a helpful error.

ducky64, 2017-02-06:
This will be implemented after / with #440

closed 2017-09-27 by ducky64