chipsalliance/chisel #401: Structural Types on IOs

ducky64, 2016-12-09:
Much module code is of the form
```
class Blah extends Module {
  val io = IO(new Bundle{
    val bundleElt = ...
  })
}
```
This tends to cause lots of `reflective access of structural type member value valid should be enabled by making the implicit value scala.language.reflectiveCalls visible.` feature warnings in sbt, the cause of which (according to [this StackOverflow post](https://stackoverflow.com/questions/26787871/warning-about-reflective-access-of-structural-type-member-in-scala)) is that the `new Bundle{...}` creates a structurally typed object.

With feature warnings now on across chisel3, this gives a lot of spam compiling chiselTests (and use of reflection generally isn't clean, though it's probably safe to assume Scala does the translations correctly). Thoughts on fixing this (style rules wise)? Options I can think of are:
- Require IO bundle definitions to be in a (possibly inner) class definition. Increases boilerplate.
- Explicitly import `scala.language.reflectiveCalls` essentially everywhere. Also boilerplate, but less of it. Would prevent this feature warning on non-Chisel-related structural types as well, which may be undesirable.
- Selectively disable this particular warning. Is this even possible?
- Disable feature warnings. Probably not the best idea.
- Something else?

sdtwigg, 2016-12-09:
You can add an option to turn on the language at the scalac level. "-language:reflectiveCalls"

It is generally advised you only turn this on for compiling Chisel Modules (which, during expected operation, are not created at anywhere near enough frequency for the performance penalty to be a problem).

If you have projects that are actually using Scala to implement some sort of 'production-level' service thought, you shouldn't use this feature or flag. (For example, you wouldn't want this used by FIRRTLs internal data structures).

jackkoenig, 2016-12-15:
I think @sdtwigg's solution is sensible, we could just add this scalac option in chisel3's build.sbt.
Regardless the 222 warnings about this that I get are super annoying

ducky64, 2016-12-15:
The solution will work but it can also cause false negatives. Since compiler flags are set at the project instead of the file level, I guess people just need to be careful about not mixing Chisel RTL and Scala-level libraries in the same project.

I also don't see a better solution, but overall I imagine structural types are also a rarity outside Module Bundles

closed 2016-12-15 by ducky64

jackkoenig, 2016-12-15:
For mixing Chisel RTL and Scala-code, we can just encourage people not to do this. In the FAQ or something we can tell them to use the import or to define explicit classes for IO.