chipsalliance/chisel #402: cloneType fails on IO Bundle with arguments

ducky64, 2016-12-09:
This test code fails with a binding error `Error: iodef.uint: Already bound to PortBinding(chiselTests.MyInnerModule@5,Some(output))`:
```
package chiselTests

import chisel3._

class MyBundle(width: Int) extends Bundle {
  val uint = Output(UInt(32.W))
}

class MyInnerModule(width: Int) extends Module {
  val io = IO(new MyBundle(width))
}

class MyOuterModule(width: Int) extends Module {
  val innerModule = Module(new MyInnerModule(width))
  val io = IO(innerModule.io.cloneType)
  io <> innerModule.io
}

class ClonedIOSpec extends ChiselPropSpec {
  property("Cloned IO should elaborate") {
    elaborate { new MyOuterModule(32) }
  }
}

```
If the argument from MyInnerIO is eliminated (note that it's not even being used right now), everything seems to elaborate fine, so it seems to be related to the presence of an argument in MyInnerBundle.

ducky64, 2016-12-09:
Upon further testing, it seems that when Bundle's default cloneType doesn't succeed, it registers a Builder error and returns the existing object. However, the resulting Binding error fails at the invocation site, throwing an exception before the Builder error is displayed.

jackkoenig, 2016-12-15:
This might be fixed by https://github.com/ucb-bar/chisel3/pull/360

closed 2017-02-06 by ducky64