chipsalliance/chisel #403: Bitwise reductions operators not available？

colin4124, 2016-12-12:
I found that Bitwise reductions operators (xorR, orR, andR) list in https://github.com/ucb-bar/chisel3/wiki/Builtin%20Operators and https://chisel.eecs.berkeley.edu/doc/chisel-cheatsheet3.pdf

But I can't use it in chisel-tutorial (ID: ca9d5618acb77330251b96cf8ea82ebf10941147 ) with `import chisel3._` and Rocket Chip (ID: 4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7) with `import Chisel._
 ` 
I also can't found in API https://chisel.eecs.berkeley.edu/api/index.html

If these are available, what should I do to use them?

colin4124, 2016-12-14:
I used `xorR(io.in0 & io.in1)` with chisel2 before, but can't work with chisel3 (id: c5b39d0)

I look at source file at `chiselFrontend/src/main/scala/chisel3/core/Bits.scala`, and I hack this way:
` (io.in0 & io.in1).xorR()`. It works!

Is it the new way to use **xorR()**?

aswaterman, 2016-12-14:
Yes, I've updated the wiki.  @ucbjrl can you update the cheat sheet to match the wiki with respect to the reduction operators?

colin4124, 2016-12-14:
@aswaterman Thinks very much！

aswaterman, 2016-12-14:
I'm going to reopen the issue to track it until the cheat sheet is updated.

ucbjrl, 2016-12-14:
@chick , I've updated the cheatsheet in pr ucb-bar/chisel-doc#2. Could you review and possibly adjust the layout (too much space between "Operators" and "Methods")?

ucbjrl, 2016-12-15:
Closed by ucb-bar/chisel-doc#2 - cheatsheet updated and pushed to web site.

closed 2016-12-15 by ucbjrl