chipsalliance/chisel #411: ScalaDoc Improvements

jackkoenig, 2016-12-14:
The chisel3 ScalaDoc needs some cleaning up.

chisel3.util (Note these are done by file in src/main/scala/chisel3/util)
- [x] Arbiter through CircuitMath - @ducky64 
- [ ] CircuitMath through ImplicitConversions - @donggyukim 
- [ ] LFSR through OneHot - @davidbiancolin 
- [ ] Reg through util - @jackkoenig

chisel3.core
- [ ] Vec - @chick 
- [ ] Bundle - @chick 

Recall the discussion in #398 that examples belong on the class if it exists, the object otherwise. On apply methods and the like, just document what the different parameters are. For inspiration, look at [Scala List](http://www.scala-lang.org/api/current/scala/collection/immutable/List.html).

Feel free to suggest more items to add.

ducky64, 2017-04-18:
*poke*

ducky64, 2019-06-10:
Skimmed through the contents of utils, and everything looks documented. Once the final fixes from #1015 are merged we can close this,

closed 2019-06-11 by ducky64