chipsalliance/chisel #413: Expose invalidate API and stop invalidating everything by default

jackkoenig, 2016-12-14:
It is safer to error when things are left unconnected. For use cases where various things are expected not to be connected, the user can explicitly invalidate Chisel *values* (not *types*).

ucbjrl, 2016-12-15:
I've started a Google Doc for the [Invalidate API](https://docs.google.com/a/berkeley.edu/document/d/1EzhXL5EMw8vQMNppmqVNJC4qVyEST7tKzW9YBNamIOo/edit?usp=sharing). Please comment or edit that document. Once we've reached an internal consensus, we'll publish the content for public comment.

colinschmidt, 2016-12-16:
Presumably this wont affect code that doesn't import chisel3._

jackkoenig, 2016-12-16:
This will not affect the compatibility wrapper. It is intended to be improved safety for users of chisel3._

ucbjrl, 2017-11-08:
Fixed by #645

closed 2017-11-08 by ucbjrl