chipsalliance/chisel #414: Chisel's IR is public

jackkoenig, 2016-12-15:
Chisel's IR (which is not Firrtl) is currently public and thus part of the API. Should this be the case? Now is the time to make it private if we're going to.

ducky64, 2016-12-15:
Agree with making it private. It's already part of the `internal` package, and we should have reserve the flexibility to change it in the near future (there was talk about making Chisel directly emit firrtl, instead of having our own emitter in Chisel). Some constructs are exposed though (like `Width`), but some refactoring can fix that.

aswaterman, 2016-12-15:
Please make sure rocket-chip's use case continues to be supported (e.g. https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/util/GeneratorUtils.scala).

jackkoenig, 2016-12-15:
A reasonable approach might be to comment very explicitly on the internal package that anything inside the package is subject to changing arbitrarily so only advanced users willing to deal with API changes should dare venture inside. Also, if we could exclude everything inside the internal package from the published ScalaDoc, then it's pretty hidden.

aswaterman, 2016-12-16:
I think this is a pragmatic compromise.

jackkoenig, 2017-11-08:
See https://github.com/freechipsproject/rocket-chip/pull/1094/files#diff-cc545926dc7cc33a4838a84a0515fe82

ducky64, 2017-12-13:
Resolution: change chisel3.internal.firrtl to chisel3.internal.ir (because its not strictly FIRRTL - Chisel has it's internal IR that's slightly different) and make it private. There are some cases where the IR is currently inadvertently exposed, these should be fixed.
People should use FIRRTL's IR instead (for AST traversals and stuff), and we should add an API to access the elaborated module (currently none of the Driver calls return the elaborated Module).