chipsalliance/chisel #416: What is the BlackBox API do for circuits with reset and clock

chick, 2016-12-15:
@grebe gas reported (and so have some NGC folks) that some tools complain about ports not being connected when using the floating point blackboxes b/c the clock and reset don't get connected. 
What is the correct/desired behavior anywhere that blackboxes have clock and reset?

jackkoenig, 2016-12-15:
My understanding is that BlackBoxes don't have implicit clock and reset.

For example:

```scala
class MyBB extends BlackBox {
  val io = IO(new Bundle {
    val a = Input(UInt(32.W))
    val b = Output(UInt(32.W))
  })
}

class BBTop extends Module {
  val io = IO(new Bundle {
    val a = Input(UInt(32.W))
    val b = Output(UInt(32.W))
  })
  val bb = Module(new MyBB)
  io <> bb.io
}
```

Results in:
```
circuit BBTop :
  extmodule MyBB :
    output b : UInt<32>
    input a : UInt<32>

    defname = MyBB

  module BBTop :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip a : UInt<32>, b : UInt<32>}

    io is invalid
    io is invalid
    inst bb of MyBB @[BB.scala 20:18]
    bb.b is invalid
    bb.a is invalid
    io.b <= bb.b @[BB.scala 21:6]
    bb.a <= io.a @[BB.scala 21:6]
```

If you have clock or reset on your BlackBox you have to add it and connect it yourself (I think). We could consider adding traits like "Clock" and "Reset" so you could write
```scala
class MyBB extends BlackBox with Clock with Reset ...
```
in order to get similar behavior to standard Chisel modules

edwardcwang, 2016-12-15:
But since BlackBoxes may be third party Verilog modules which one has no control over, they might name their reset and clock pins differently. I think one'd have to do something like

```scala
val clk = Input(Clock())
val rst = Input(Bool())
```

Unless one wants to go and rewrite all the references to clk in the Verilog itself.

grebe, 2016-12-15:
I think the current behavior is very reasonable, although I think it is a bit confusing that BlackBox extends Module b/c I would expect something that extends module to behave like a Module and add clock and reset. There is a TODO floating around [in the source](https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/BlackBox.scala#L34) saying that BlackBox should extend BaseModule instead- I think that would help. A wiki page documenting how blackboxes are meant to be used would help too.

jackkoenig, 2016-12-15:
Totally agree, I believe fixing the Module/BlackBox hierarchy is in the works, stay tuned.
Also, as always, we do need better documentation.

ducky64, 2017-02-06:
Superseded by #427

closed 2017-02-06 by ducky64