chipsalliance/chisel #419: cloneType vs. chiselCloneType

ducky64, 2016-12-16:
cloneType doesn't seem to preserve IO directionality, while chiselCloneType does. However, the names are highly unintuitive. 

Related to #368, it might be a good idea to have a `.clone` (which preserves directionality) and `.asType` / `.getType` which doesn't. (and in my opinion, the behavior is clearer too)

jackkoenig, 2016-12-16:
I agree that `.cloneType` vs `.chiselCloneType` is confusing. I don't think that `.clone` is a good candidate because of it's Java meaning and all of the baggage that comes with that, but I do think that `.asType` or `.getType` are good names. Picking these names is pretty tough though, and it might be wise to drop `.cloneType` unless it does the exact same thing it does now or did in the past...

ducky64, 2016-12-16:
If we go with the stricter type system in #368 I can see there being a use for two types of clone, one that preserves directionality / "bindings" (I've found it useful to be able to clone another block's IO) and another that does not (for instance, to use a IO data type in a Reg).

aswaterman, 2016-12-16:
Do avoid the name clone. That was the chisel1 name for cloneType and it was problematic for the reason @jackkoenig mentioned.

ucbjrl, 2016-12-16:
@sdtwigg used `newType` in the original #200 pr. My renaming that to `chiselCloneType`, although well-intentioned, was probably a mistake: I wanted to mark it _for internal use only_.

Do we need three type generators:
- `newUnboundType` (corresponding to the current `cloneType`
- `newUnboundTypeWithWidth` (corresponding to the `cloneTypeWithWidth`
- `newBoundType` (corresponding to the current `chiselCloneType`

I'm hoping we don't need to expose all three to user code. `cloneType` exists only because we can't reproduce the client's type construction environment without help from the client. Since it's now part of the public API (and we don't intend to change its semantics), it should probably retain its current name. I'm quite happy changing the name of the other two, and possibly using `newUboundType` internally, passing the buck to `cloneType` to do the heavy lifting.

ducky64, 2016-12-16:
Is there ever a case where we want clone type while dropping widths? While widths aren't part of the types as far as the type system is concerned, I think that's just an implementation detail.

Also, do we want to expose Binding as something users should know about? If not, then it shouldn't be in names of API functions. I also think it makes sense to describe APIs in terms of how it affects the user. Something like `.asType()` / `.getType()` (current `cloneTypeWidth`) vs. like `.cloneIo()` (current `chiselCloneType`, or whatever the full scope of Bindings is).

ducky64, 2017-01-19:
Resolution as follows:
- Deprecate chiselCloneType
- Deprecate cloneType as a user callable function. Eventually this may be made protected. It will still remain as a function that users need to overload in some cases, like a Bundle with constructor arguments.
- Introduce `getType`, which can only be called on a terminally-bound type (something that was wrapped with Wire(...) or IO(...)) and returns the non-terminally-bound version. It is an error to invoke this on a non-terminally-bound type (with the overall philosophy of stricter separation of model nodes and hardware graph nodes)
- Introduce `wipeDir`, which an only be called on a non-terminally-bound type. It does exactly what you think it should do, it wipes the direction so you can instantiate a Reg or Mem of a bidirectional Bundle. 
  - It is an error to invoke this on a terminally-bound type because you almost always want to use `getType` instead. If you want to do this, you have to go the extra step and use `getType.wipeDir`
- Introduce requireIsHardware (or requireUnbound, if Bindings is more than an implementation detail?), requireDirectioned, requireUndirectioned APIs. Please suggest better names! This helps library writers enforce the strict separation of model vs. hardware graph types.
- Ideally, all this would be done in the (Scala) type system, but that would be a massive structural, and most likely very backwards incompatible, change in Chisel. So impractical. But I'd love to be proven wrong here.

Additionally, cloneType's contract should be such that it must return a non-terminally bound type, so terminal bindings are always final (assign once, non mutable after).
wipeDir will mutably clear non-terminal direction bindings using a special API. Once #440 is implemented, bindings can be made write-once with the exception of wipeDir, which will have access to a private[chisel3] API. It's not possible to clone without bindings, since directionality may be bound in Bundle constructors.

ducky64, 2017-01-20:
Should address #445, #368. Implementation depends on #440.

closed 2017-11-08 by ducky64