chipsalliance/chisel #421: Does fromBits work?

grebe, 2016-12-21:
It is looking like my fix for FixedPoint.fromBits is imperfect (see #391). In trying to figure out how everything works, I wrote this test:

```
class SIntFromBitsTester extends BasicTester {
  val sint = SInt(2.W)
  val uint = 3.U
  val b: SInt = sint.fromBits(uint)
  val sint_answer = (-1).S(2.W)
  
  assert(b === sint_answer)
  stop()
}
```

This fails with `firrtl.passes.CheckTypes$InvalidConnect: Type mismatch. Cannot connect b to UInt<2>("h3")` because the emitted firrtl tries to directly connect a UInt to an SInt. The offending lines of emitted firrtl are:

```
    wire b : SInt<2>
    b is invalid
    b <= UInt<2>("h03")
```

Am I wrong to think the test should work as written? It seems fromBits will only work for Bundles/Vectors entirely composed of UInts.

ducky64, 2016-12-22:
Seems like SInt doesn't overload fromBits, and it's defaulting to the Bits.fromBits implementation which doesn't do anything special in terms of FIRRTL generation.

Side notes: not sure why Data has a base fromBits implementation (the current one should probably to into Aggregate/Bundle/Vec, especially since Bits overrides it completely). We might also need more SInt in the stock Chisel3 test suite...

grebe, 2016-12-22:
It is true that sint doesn't overload frombits, but even if it did there
would be a problem. A bundle with an sint would give the same error because
frombits doesn't recursively call itself, it just bitslices and makes the
assignment directly. I feel like frombits should call itself recursively,
but it's also possible to fix this in the connect operator

<https://github.com/ucb-bar/chisel3/issues/421#issuecomment-268728662>, or mute
the thread

ducky64, 2017-02-06:
Appears fixed?

closed 2017-02-06 by ducky64