chipsalliance/chisel #425: PeekPokeTester not taking values beyond 32 bits

preyas3359, 2016-12-31:
Hi,

I am seeing 2 issues with FIRRTL while using PeekPokeTester:

1. For hex numbers FIRRTL is converting to signed integer and not unsigned integers.
2. For more than 32 bits signal, only last 32 bits are taken into consideration by FIRRTL. 

For example, my current test case is simple i/p directly connected to o/p. (both i/p and o/p are defined as 64 bits UInt)

Test 1: (works fine)
poke (c.io.A, 0x0000000070000000L)
expect(c.io.Out, 0x0000000070000000L)

Test 2:(Test Fails)
poke (c.io.A, 0x0000000770000000L)
expect (c.io.Out, 0x0000000770000000L)

Output only takes value of last 32 bits (70000000) and test fails.

Test 3:(FIRRTL generates an error)
poke (c.io.A, 0x0000000080000000L)
expect (c.io.Out, 0x0000000080000000L)

Compilation error from FIRRTL: 
Error: poke(io_A, -2147483648)
firrtl_interpreter.InterpreterException: error: ConcreteUInt(-2147483648, 64) bad value -2147483648 must be >= 0

Is this behavior expected? Or am I doing something wrong?
Can somebody please help?

Regards,
Preyas

chick, 2017-01-03:
I am looking into this, can you please add a gist of the code that exhibits the error. 
Thanks, chick

preyas3359, 2017-01-03:
Hi Chick,

Cleaning up current code and putting it here will take some time. But, below mentioned simple test case also exhibits same issue. If you still need the actual code, I can post it by sometime tomorrow.

[Test.txt](https://github.com/ucb-bar/chisel3/files/681745/Test.txt)
[Test_tb.txt](https://github.com/ucb-bar/chisel3/files/681746/Test_tb.txt)

Thanks & Regards,
Preyas

chick, 2017-01-03:
Hi Preyas,
I have pushed a fix in [/ucb-bar/chisel-testers/pull/76](https://github.com/ucb-bar/chisel-testers/pull/76), that I think addresses the issue. It includes a test very similar to yours.
Can you possibly test with that branch?
-chick

chick, 2017-01-03:
The PR has been merged so can now be found on master branch of chisel-testers

preyas3359, 2017-01-03:
Hi Chick,

Thanks. I will check and close the issue.

Regards,
Preyas

preyas3359, 2017-01-04:
Hi Chick,

I was able to work with recent release. Initial issue #2 is solved now. But below mentioned issue still persists. Unfortunately, none of the tests I posted in issue covered this.

2) For hex numbers FIRRTL is converting to signed integer and not unsigned integers.

Below mention test will cover it. (it fails with FIRRTL generating signed integer)

Test 3:(FIRRTL generates an error)
poke (c.io.A, 0x8000000080000000L)
expect (c.io.Out, 0x8000000080000000L)

There is a workaround in test case definition which can solve this.

def asUnsigned(unsignedLong: Long) =
       (BigInt(unsignedLong >>> 1) << 1) + (unsignedLong & 1)
 
val a : Long = 0x8000000080000000L
poke(c.io.A, asUnsigned(a))
step(1)
expect(c.io.Out, asUnsigned(a))

I think this kind of conversion should be part of tester code. Kindly let me know your views on it.

Thanks and Regards,
Preyas

chick, 2017-01-04:
I think the following is an easier workaround.

poke (c.io.in, BigInt("8000000080000000", 16))
expect (c.io.out, BigInt("8000000080000000", 16))

Does this work for you?

Chick Markley
Staff Programmer -- Aspire Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu

preyas3359, 2017-01-05:
Hi Chick,

It does work for me. Thanks. 

Regards,
Preyas

closed 2017-01-05 by chick