chipsalliance/chisel #427: Special Top Module class to rename/delete clock/reset

Martoni, 2017-01-02:
It seems to be impossible to delete or rename clock and reset signals of Module declared with Chisel3. But it can be useful for some design where clock come from differential input buffer for example.

I had this problem with a Xilinx FPGA Kit [where a special buffer must be instantiated to use clock](http://stackoverflow.com/questions/41427717/how-to-delete-clock-signal-on-chisel3-top-module/41430368#41430368).

It could be useful to have a special top module where we can decide if we use clock and reset and where we can rename io signal. Like BlackBox.

ben-k, 2017-01-09:
I'll second this - it currently seems to require external post-processing of the generated verilog to address, and seems like it will be a very common issue in non-toy designs.

It might be better addressed in firrtl, though.

edwardcwang, 2017-01-09:
Another alternative: wrap the generated 'top-level' Verilog into another Verilog file and just connect the wires.

sdtwigg, 2017-01-09:
We addressed this by redoing the Module hierarchy (BlackBox already was working as a hack so this was a good opportunity for reconciliation)

The hierarchy is:
BaseModule
BaseModule -> BlackBox (user-extendable)
BaseModule -> UserDefinedModule
UserDefinedModule -> ImplicitModule (user-extendable, Module is type aliased to this)
UserDefinedModule -> RawModule

A BaseModule is basically any module that can be instantiated as a submodule. Thus, all it has is an io. (Note that the 'io' field might not be the true full io)
A UserDefinedModule is a module that actually holds an implementation in Chisel. Thus, it has both an io and commands. It can also have submodules/definitions and therefor all 'parent' fields are actually UserDefinedModule

ImplicitModule is an old Chisel.Module and thus has a clock/reset pin and the io is actually put in an "io" subfield of the true io. The Module wrapper handled propagating clock/reset from ImplicitModule parent to ImplicitModule child.
RawModule is new and has no clock/reset pins. The io is the true io (excepting any ports added reflectively). No automatic propagation is done into this module or from this module into its children. Thus, an ImplicitModule in a RawModule must have its clock and reset manually connected.

This is the rough summary of the changes. There are also pending fix-ups to make the clock propagation for ImplicitModule actually use scala implicits (likely tied to the Module wrapper call to avoid excessive API churn) so that it is easier for libraries to write functions that instantiate (and hookup) ImplicitModules which can be used in either ImplicitModule or RawModule. Also, implicits are a little easier to track because they are lexically resolved (rather than dynamically).

Martoni, 2017-01-10:
@sdtwigg, do you have a patch for your proposal ? I can test it if needed.

sdtwigg, 2017-01-10:
Unfortunately no, I don't have the time (or will) to deal with getting this to work via the compatibility layer.

Martoni, 2017-01-11:
@sdtwigg, but can you share the code you have ? I'm interested by it to test. Thx

shunshou, 2017-01-15:
I think whatever @ducky64 did with JTAG should be a somewhat sufficient hack to get what you want?   You definitely need a wrapper module in Chisel though, which is really annoying... i.e. something like Top level module with normal IO that contains yourSpecificClk (Bool). yourSpecificClk then gets passed down to the relevant submodule [what would've been the top if you could just rename your module clk...] as that module's override_clock. It's probably fair that each module has only 1 clock... Apparently you can test this via Verilator too? At the expense of increased simulation time... i.e. poking yourSpecificClk every n steps of the Chisel clk...

That's also where I'd really like to not limit IO to just 1 bundle... I wish we could have a normal IO and an clk/reset override IO or some other way to make overrides less intrusive until a more formalized solution is put in place...

ducky64, 2017-01-15:
So that solution addresses how to add extra pins, but doesn't allow you to delete the built-in clock and reset pins. So the JTAG FPGA examples all have a clock and reset line on their top module, and it seems that the synthesis tool doesn't complain that the top-level reset pin doesn't actually go anywhere...

@sdtwigg if you can share code (or a more well-defined API spec) then we can try to implement it and deal with possibly legacy issues via the compatibility layer.

shunshou, 2017-01-15:
I think deleting built-in clock/reset is a *very useful* luxury. It's just like why I wanted Optionable IO's before... Tools might get upset about unused IO but at least they won't fail...

ducky64, 2017-01-19:
We will implement as follows, based on our understand of Stephen's method:
BaseModule: no contents, instantiable
BlackBox extends BaseModule
UserDefinedModule extends BaseModule: this module can contain Chisel RTL. No default clock or reset lines. No default IO. User should be able to specify non-`io` ports, ideally multiple of them.
ImplicitModule extends UserModule: has clock, reset, and io, essentially current Chisel Module.
RawModule: will be the user-facing version of UserDefinedModule
Module: type-aliases to ImplicitModule, the user-facing version of ImplicitModule.

Hopefully implementation to come within two weeks.

ducky64, 2017-01-31:
So really fun problem:
To implement multiple IOs, I have IO(...) register the port as an IO.
Now, this becomes problematic if you override the value later. This actually happens in the cookbook, where the BasicTester defines the tester communication Bundle, then the CookbookTester extends BasicTester, but overloads `io` to be an empty Bundle.

I think the solution here is "don't override your ios", but I do want to see what everyone else thinks, and whether there's ever a good reason have overriden Chisel RTL.

aswaterman, 2017-01-31:
We use overrides from time to time when similar modules differ only slightly.  It is weird to take the DSL approach of making Modules classes, then disallow basic inheritance features.

ducky64, 2017-01-31:
So problem is that the overridden code seems to be running and calling Chisel commands that are generating nodes on the hardware graph...

shunshou, 2017-03-18:
@ducky64 is there any chance RawModule or similar can be added to experimental or something in the next <1 week? I have a lot of clock stuff that I'm afraid I might've screwed up, but it's not easy sanity checking whether Regs are getting the right clks when I'm still trying to run my peek/poke tests with steps off of the default clk. It'd be nice if only my top module used the default clock/reset and all the sub modules could just be without clock/reset to make sure I didn't wire stuff up incorrectly.

jackkoenig, 2017-03-18:
For sanity checking you can just wire the default clocks of your submodules
to false

shunshou, 2017-03-18:
Sanity checking makes more sense at the module level rather than the instance level -- in case you missed an instance clock?

jackkoenig, 2017-03-18:
You could try something like:

```scala
class WouldPreferToExtendRawModule extends Module { withClock(false.B.asClock) {
  // ...
} }
```

Anyway these are just ideas for a stopgap in case RawModule won't be ready quickly enough

ducky64, 2017-03-18:
@shunshou so the PR (#469) is ready but (still) waiting review. As the RawModule addition requires significant refactors in core chisel3, it's unlikely it could be split out...

ducky64, 2017-04-15:
Merged!

closed 2017-04-15 by ducky64