chipsalliance/chisel #428: How to print message to different files during simulation?

sashimi-yzh, 2017-01-03:
I want to output the value of different signals to different files in verilator. First I try
```
System.err.printf("aaa\n")
```
But it will output during the generation of fir code, which is not my expectation. Actually, I want something like
```
fprintf(my_file,  "valid = %x\n", valid)
```
in chisel.

I wonder how to do this. Thanks!

edwardcwang, 2017-01-04:
Something along the lines of `fprintf(my_file,  "valid = %x\n", peek(c.io.valid))`?

closed 2017-01-04 by edwardcwang

sashimi-yzh, 2017-01-05:
I know `printf()` can output message during the FIRRTL simulation. But I want to let different `printf()` output to different files, instead of stdout.

edwardcwang, 2017-01-11:
Is it not possible to just use the Java file I/O APIs in that case in lieu of System.out.printf()? https://www.tutorialspoint.com/scala/scala_file_io.htm

jackkoenig, 2017-01-11:
Java file I/O occurs at elaboration time, the desired feature is for
printing during simulation. A proper logging framework for Chisel is a
desired feature but not yet implemented.

sashimi-yzh, 2017-01-12:
@edwardcwang As jackkoenig says, Java file I/O will output message at elaboration, as outputting the configuration string.

@jackkoenig Thank you for the reply! It seems that there does not exist a solution to this issue at this time.