chipsalliance/chisel #435: How to print different message to different file during simulation?

sashimi-yzh, 2017-01-11:
I have added the UART scala code from the sifive repo to the rocket-chip. In the code, [there is a `printf()` to output the sending char during simulation](https://github.com/sifive/sifive-blocks/blob/master/src/main/scala/devices/uart/UART.scala#L79):
```
printf("%c", io.in.bits)
```
At the same time, I hope that [the rocket core will output execution logs](https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/rocket/rocket.scala#L687) to help me for debugging:
```
printf("C%d: %d [%d] pc=[%x] W[r%d=%x][%d] R[r%d=%x] R[r%d=%x] inst=[%x] DASM(%x)\n",
...
```

What troubles me is that these two `printf()` will output to screen at the same time during simulation. This makes messages sent by UART interrupted by a large number of logs. I hope the generated code can separate the output of these two `printf()`. For example, the execution logs will be printed to a file `log.txt`, while the UART char is still output to the screen. What should I do?

In fact, I opened an issue #428 for the same trouble last week, but it was misunderstood and closed, and I have not got the solution yet. I hope I have made this issue clear this time. Thanks!

seldridge, 2017-01-11:
FWIW, a non-solution is to just use a disciplined printf format for what you want to add.

All the rocket log printfs start with `C` so just inverting a match on that will dump everything else.
```
grep -v ^C rocket.log > uart.txt
grep    ^C rocket.log > rocket.txt
```
This can obviously get overloaded if you have other printfs that start with `C`, but as long as you're in control of the printfs (e.g., `printf("[sashimi-yzh] ...", ...)` you can use a unique printf that won't conflict with rocket.

sashimi-yzh, 2017-01-12:
@seldridge As discussed in issue #428, there does not exist a solution at this time. So thank you for the non-solution!

ducky64, 2017-04-18:
Ultimately, this is a request for a logging library that dumps prints in simulation.
Renamed issue.