chipsalliance/chisel #437: Are `EnqIO` and `DeqIO` backwards?

albert-magyar, 2017-01-11:
I see that `EnqIO` is used on the _output_ of `Queue` and vice-versa. Is this intentional? Chisel2 did not use `EnqIO` or `DeqIO` in `Queue` at all (as far as I can tell).

https://github.com/ucb-bar/chisel3/blob/3215df07a97b58babb1deb3fab0928198b1daad2/src/main/scala/chisel3/util/Decoupled.scala#L131-L146

aswaterman, 2017-01-11:
I thought we were removing them altogether: https://github.com/ucb-bar/chisel3/pull/322

albert-magyar, 2017-01-11:
That would make even more sense. @ucbjrl, is there a reason #322 was reverted?

ucbjrl, 2017-01-12:
According to the commit comment:

"Update chisel-testers before commiting this change (deleting EnqIO/DeqIO)."

ucbjrl, 2017-01-12:
W.r.t. the apparent reversal of EnqIO/DeqIO, I believe I managed to convince myself that inside the Queue, they are inverse of what the client sees from outside (if that makes sense).

aswaterman, 2017-01-12:
Ah, I see.  Removing them still strikes me as the right thing to do.  This conversation (and Stephen's decision to change them) is compelling evidence that they are confusing, because whether they are named correctly is context/orientation-dependent.  I don't know what the chisel-testers dependence was referring to, but hopefully that change is just a search and replace job.  @chick ?

jackkoenig, 2017-01-12:
When I first ran into them I had the same confusion as Albert. I'm inclined
to agree that they're more trouble than they're worth.

ucbjrl, 2017-01-12:
I started the replacement, then realized I was basically replacing EnqIO/DeqIO with EnqIO/DeqIO and gave up. There is obviously a use case for them. @chick and I will discuss the issue.

ucbjrl, 2017-02-07:
I think one source of confusion is the comment for ReadyValid:

```
/** An I/O Bundle containing 'valid' and 'ready' signals that handshake
  * the transfer of data stored in the 'bits' subfield.
  * The base protocol implied by the directionality is that the consumer
  * uses the flipped interface. Actual semantics of ready/valid are
  * enforced via use of concrete subclasses.
  */
abstract class ReadyValidIO[+T <: Data](gen: T) extends Bundle
{
  val ready = Input(Bool())
  val valid = Output(Bool())
  val bits  = Output(gen.chiselCloneType)
}
```

It sure looks to me that this interface is defined from the perspective of the consumer (`bits` are `Output`). I think the comment should be inverted:

```
  * The base protocol implied by the directionality is that the producer
  * uses the flipped interface. Actual semantics of ready/valid are
```

mwachs5, 2017-02-07:
What? Wouldn't the consumer have the bits as input? For your Queue case your confusion may be justified, but ReadyValidIO and the comments seem just fine in the general case.

I was taught that interfaces should be conventionally defined in the "master" way, and "slave" uses flipped version. The current ReadyValidIO code and comments above seems consistent with that...

ucbjrl, 2017-02-07:
You are correct. My confusion stems from the fact that the interface is defined from the point of view of something the consumer can use (consume). So of course, something providing a ReadyValid interface for the consumer, must present `bits` as `Output` so the consumer can connect them to the appropriate `Input` in their circuit.

I retract my suggestion to invert the comment.

edwardcwang, 2017-02-07:
What about a comment along the lines of "the producer uses the interface as-is (outputs bits) while the consumer uses the flipped interface (input bits)" to fully clarify the confusion?

ducky64, 2017-04-18:
Solution: make Decoupled and queues use DecoupledIOs, and clean up everything that uses it like chisel-testers. Unlikely to break rocket-chip.

commit cb28230d (2017-05-19): Update comments describing Decoupled/ReadyValid - fix #437. (#493)

* Update comments describing Decoupled/ReadyValid.
It seems there is a valid use case for EnqIO/DeqIO and updating the comments may clear some of the confusion and encourage their usage.

* Update comments - no functional changes.
Re-flow comments for ReadyValidIO()
Add gen param to DecoupledIO() and IrrevocableIO().

* Update code and comment now that #492 is resolved
closed 2017-05-19 by ucbjrl