chipsalliance/chisel #438: Reg + Type parameterization

shunshou, 2017-01-15:
Ran into 

```

inferred type arguments [Object] do not conform to method apply's type parameter bounds [T <: chisel3.core.Data]
[error]   io.out.bits.y := Reg(next = io.in.bits.y)

```

error, which this comment in the code alluded to [assuming you happened to see the comment]:

```
@note this may result in a type error if called from a type parameterized
    * function, since the Scala compiler isn't smart enough to know that null
    * is a valid value. In those cases, you can either use the outType only Reg
    * constructor or pass in `null.asInstanceOf[T]`.
    */
```

Can you not just make `def apply[T <: Data](t: T = null, next: T = null, init: T = null)` private and case out all of the possibilities via RegNext, RegInit, RegNextWithInit (doesn't exist), etc. to save people the trouble of debugging stuff like this?

ducky64, 2017-01-15:
I think that's a valid possible solution to the Reg problem (eliminate the nulls, which are *terrible*) by having only a single-argument Reg constructor (analogous to Wire, which only creates a synthesizable node), and requiring the use of RegNext and friends for the next=... and init=... arguments.

This would be an API changing thing (so it's a good thing 3.0.0 is still on the horizon), and the unclean Reg could stay in compatibility layer for those stuck at 2.x semantics.

(side note: many of these notes documenting non-intuitive behavior was from the initial push to 3.0, where we prioritized API compatibility and avoided solutions which could cause existing code to break. The compatibility layer makes that less necessary now.)

Thoughts? @sdtwigg (you must hate the use of nulls, did you have a solution?), @azidar, @jackkoenig ?

shunshou, 2017-01-15:
@sdtwigg was the one that reminded me of RegNext. :P 

Regarding the notes: I assume the errors you'd run into would basically be the same each time. It'd be much more useful to create some searchable list of errors and solutions i.e.

Search "inferred type arguments [Object] do not conform to method apply's type parameter bounds [T <: chisel3.core.Data]", "Reg"

Result: Don't use Reg(next = blah)

Rather than making people scratch their heads... Maybe people would think to search for Reg specifically; maybe they'd think they just wrote their code incorrectly (esp. for people who are new to type parameterization)... but that just costs design time...

ducky64, 2017-01-19:
We will (try to) deprecate the optional `init` and `next` parameters from `Reg(...)`. Reg will now only take a type (the first argument) and will behave similar to Wire(...) - it creates a synthesizable node without any assignments.

The next/init functionality will be replaced by the (existing) RegNext/RegInit.

Side note: we considered flipping arguments in RegEnable to be `RegEnable(next, enable, init)` instead of `RegEnable(next, init, enable)` (where the `init` is inexplicably dropped between `next` and `enable` which are conceptually more related) though it's hard to do this with a deprecation phase. Does anyone have stronger opinions about argument order?

mwachs5, 2017-01-27:
So is there a RegNextInitEnable...? and all the combinations?

shunshou, 2017-01-27:
There *needs* to be if there isn't...

ducky64, 2017-01-27:
I don't think we've ever had this.
The current way I'm going it that RegInit and RegNext are part of chisel core, but RegEnable is still a util. If you needed RegNextInitEnable, you should use RegInit, then specify the update on a separate line.

Imagine how many arguments you would need for a RegNextInitEnable, and how difficult it would be to keep track of all of them...

mwachs5, 2017-01-27:
The common case is a register with init and next. So what is the way to do that one now?

ducky64, 2017-01-27:
RegNext(next, init)

shunshou, 2017-01-27:
Is this going to change?

```
object RegNext {
  /** Returns a register with the specified next and no reset initialization.
    *
    * Essentially a 1-cycle delayed version of the input signal.
    */
  def apply[T <: Data](next: T): T = Reg[T](null.asInstanceOf[T], next, null.asInstanceOf[T])

  /** Returns a register with the specified next and reset initialization.
    *
    * Essentially a 1-cycle delayed version of the input signal.
    */
  def apply[T <: Data](next: T, init: T): T = Reg[T](null.asInstanceOf[T], next, init)
}
```

Specify Type and Init in RegNext. 
Granted, if Type requires more bits than an Init Lit, it would suck if the output ended up having the same # of bits as Init. This should be double checked.

ducky64, 2017-01-27:
Those APIs will not change. (unless we want to move to Reg.next or whatever)

mwachs5, 2017-01-28:
Ok, sounds good. I never even knew about these things.

shunshou, 2017-01-28:
That's also a big problem. For Chisel2, I *really* appreciated the fact that people put together a "Chisel Cheatsheet," and used that extensively to the point where I pinned it at my desk. The fact that you need to basically dig through the code base (or ScalaDoc) to see if a common function you might want exists is troubling... That leads to a lot of code duplication in projects...

chick, 2017-01-28:
There is a [chisel3 cheatsheet](https://chisel.eecs.berkeley.edu/doc/chisel-cheatsheet3.pdf) but it needs to be updated with these latest API changes.  But please use it, that will force us to keep it up to date

ducky64, 2017-01-28:
I think a lot of functions were either outside the scope of the cheatsheet or were added after the initial version of the cheatsheet. 
We will need to have a push for updated docs sometime when 3.0.0 stabilizes

shunshou, 2017-01-28:
Please advertise that everywhere. It's, in many ways, more useful than ScalaDocs/Wiki.

grebe, 2017-01-28:
Where does the source for the cheatsheet live?

shunshou, 2017-01-28:
Also, isn't is super dated? Since that's not even the proper way to construct a wire...

ducky64, 2017-04-12:
Reg(..., init=..., next=...) deprecated in favor of RegInit/RegNext/etc which does have nulls that cause unhelpful error messages.

closed 2017-04-12 by ducky64