chipsalliance/chisel #441: Document which way Bundles are packed

mwachs5, 2017-01-18:
When turning Bundles into UInt, and vice versa, I always forget which way they are packed (what is the endianness basically). Is this documented somewhere? I would add it to the Cookbook item related to converting between Bundles and UInt, but not sure how to suggest edits to the Wiki.

jackkoenig, 2017-01-18:
I'm working on this as suggested on https://github.com/ucb-bar/chisel3/pull/366#discussion_r91828211.

For the Cookbook, working examples added [here](https://github.com/ucb-bar/chisel3/tree/master/src/test/scala/cookbook) are appreciated. I'm also working on a mirrored Github repo of the wiki to allow PRs against it directly.

mwachs5, 2017-01-18:
Ah, I guess I should have just looked at the examples more closely. Thx

closed 2017-01-18 by mwachs5