chipsalliance/chisel #442: Associate resources with Modules / etc

ducky64, 2017-01-18:
Generalization of ucb-bar/dsptools#21

BlackBoxes require additional Verilog / etc files to work, and those currently need to be tracked manually (outside Chisel, and kind of breaks the clean library abstraction). A clean solution would be to allow the Verilog resources to be complied into the JAR, and have those be copied to the output directory if the corresponding BlackBox module is used.

This would vastly improve the user experience for re-usability of BlackBox-dependent libraries.

aswaterman, 2017-01-18:
This has potential to improve usability for static BlackBox implementations, but bear in mind two caveats: BlackBox implementations can be codegen'd as a result of elaborating a Chisel design; and the JAR approach is of little help when using third-party tools and build systems.  So the current do-nothing approach should remain supported.

ducky64, 2017-01-19:
Is the codegen specified in Scala? If so, then it might make sense to have a more generalized version of this that could have a code generator associated with a Module (or perhaps any generalized object in the hardware graph), with static Verilog files being one special case.

Though for arbitrar generated code there would need to be a good deduplication strategy. For Modules, this could be the Module name (and maybe contents, as a sanity check?), since if those are conflicting the Verilog compiler is going to have a bad day anyways.

aswaterman, 2017-01-19:
Often it cannot be expressed in Scala.  The most common use case is to invoke a third-party tool, which itself is a black box in the software engineering sense.

Not sure how the deduplication interacts with parameterized BlackBoxes, but certainly BlackBoxes with the same module name and different parameters should not be deduplicated.

ducky64, 2017-02-06:
This is done!

closed 2017-02-06 by ducky64